paukstelis/octoprint_deploy #117: Octoprint failed to start after Ubuntu OS upgraded

chiksum, 2023-08-30:
Here is the diagnostic output from octoprint_deploy.sh

`octoprint_deploy diagnostic information. Please provide ALL output for support help
**************************************
/etc/octoprint_deploy
**************************************
streamer: none
octoexec: /home/chiksum/OctoPrint/bin/octoprint
octopip: /home/chiksum/OctoPrint/bin/pip
**************************************
/etc/octoprint_instances
**************************************
instance:i3megas port:5000 udev:false
cat: /etc/octoprint_cameras: No such file or directory
**************************************
/etc/octoprint_cameras
**************************************
**************************************
/etc/udev/rules.d/99-octoprint.rules
**************************************
lrwxrwxrwx 1 root root 7 Aug 30 13:52 /dev/octo_i3mega_2 -> ttyUSB0
ls: cannot access '/dev/cam*': No such file or directory
**************************************
× i3megas.service - The snappy web interface for your 3D printer
     Loaded: loaded (/etc/systemd/system/i3megas.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2023-08-30 13:49:14 UTC; 6min ago
   Main PID: 639 (code=exited, status=1/FAILURE)
        CPU: 16ms

Aug 30 13:49:14 octoprint systemd[1]: Started The snappy web interface for your 3D printer.
Aug 30 13:49:14 octoprint octoprint[639]: Traceback (most recent call last):
Aug 30 13:49:14 octoprint octoprint[639]:   File "/home/chiksum/OctoPrint/bin/octoprint", line 5, in <module>
Aug 30 13:49:14 octoprint octoprint[639]:     from octoprint import main
Aug 30 13:49:14 octoprint octoprint[639]: ModuleNotFoundError: No module named 'octoprint'
Aug 30 13:49:14 octoprint systemd[1]: i3megas.service: Main process exited, code=exited, status=1/FAILURE
Aug 30 13:49:14 octoprint systemd[1]: i3megas.service: Failed with result 'exit-code'.
Traceback (most recent call last):
  File "/home/chiksum/OctoPrint/bin/octoprint", line 5, in <module>
    from octoprint import main
ModuleNotFoundError: No module named 'octoprint'
Traceback (most recent call last):
  File "/home/chiksum/OctoPrint/bin/octoprint", line 5, in <module>
    from octoprint import main
ModuleNotFoundError: No module named 'octoprint'
Traceback (most recent call last):
  File "/home/chiksum/OctoPrint/bin/octoprint", line 5, in <module>
    from octoprint import main
ModuleNotFoundError: No module named 'octoprint'
cat: /etc/octoprint_cameras: No such file or directory
**************************************
○ haproxy.service - HAProxy Load Balancer
     Loaded: loaded (/lib/systemd/system/haproxy.service; disabled; vendor preset: enabled)
     Active: inactive (dead)
       Docs: man:haproxy(1)
             file:/usr/share/doc/haproxy/configuration.txt.gz
`

chiksum, 2023-08-30:
Great, after reinstall, octoprint is working again.
You may consider incorporating this command into the script
Thank you so much!

chiksum, 2023-09-15:
Hello, today when I try to execute `~/OctoPrint/bin/pip install --force-reinstall --no-cache-dir octoprint`
I got the following error:

> Traceback (most recent call last):
>   File "/home/chiksum/OctoPrint/bin/pip", line 5, in <module>
>     from pip._internal.cli.main import main
> ModuleNotFoundError: No module named 'pip'

paukstelis, 2023-09-15:
First, explain why you need to run the command again. You should not have to do that.

chiksum, 2023-09-15:
Because I accidentally rolled back to a previous version of Ubuntu, so I need to run the command again.

paukstelis, 2023-09-15:
It becomes very difficult to help troubleshoot when various system things are changing.
The best suggestion I have is to create a backup (you can use octoprint_deploy for this), remove everything related to octoprint_deploy (sudo octoprint_deploy/octoprint_deploy.sh remove) and then reinstall. Create an instance with the same name as your old instance, then use the restore function (tar.gz file has to be in $HOME) to copy everything back to how it was.

chiksum, 2023-09-15:
Will the restore function from octoprint_deploy restore octoprint plugin?
After restoring my instance, the settings of OctoPrint are loaded, but all of my plugins are gone.

paukstelis, 2023-09-15:
No, you will have to reinstall plugins (but the settings will be restored once they are installed).

chiksum, 2023-09-15:
Got it! Thank you!
I found out that I can list all my plugins in the 'plugins_list' file and install all recommand plugin after Prepare System, then use the restore function with octoprint_deploy.
However, this method will install plugin to all instance.

closed 2023-09-15 by chiksum