paukstelis/octoprint_deploy #120: octoprint-deploy using unavailable sudo option --basedir

zodeeeac, 2023-09-13:
Hi,

during creation of a new instance on Debian 12 I now get `sudo: unrecognized option '--basedir'`
Looks like the --basedir option on sudo does no longer exist and is still being used by octoprint-deploy ?

This is the output I get:

```shell
localadm@octoprint:~$ sudo octoprint_deploy/octoprint_deploy.sh

*************************
octoprint_deploy 1.0.4
*************************

1) Add instance	    3) Add Camera	5) Utilities	    7) Update
2) Delete instance  4) Delete Camera	6) Backup Menu	    8) Quit
Select operation: 1
Enter the name for new printer/instance (no spaces):
biqu-b1

Using a template instance allows you to copy config settings
and gcode files from one instance to your new instance.
Use an existing instance as a template? [y/n]: n
Ready to begin instance creation? [y/n]: y
Selected port is: 5000
Your new OctoPrint instance will be installed at /home/localadm/.biqu-b1

Printer auto-detection must be completed before an instance can be created.
Begin printer auto-detection for udev entry? [y/n]: y

Plug your printer in via USB now (detection time-out in 1 min)
Serial number detected as: 1003202C95DF80A962AD7186F50020C6
No duplicate serial number detected.

Would you like to auto detect an associated USB camera (experimental)? [y/n]: n
Ready to write all changes. Do you want to proceed? [y/n]: y
Uniquifying instance...
sudo: unrecognized option '--basedir'
usage: sudo -h | -K | -k | -V
usage: sudo -v [-ABkNnS] [-g group] [-h host] [-p prompt] [-u user]
usage: sudo -l [-ABkNnS] [-g group] [-h host] [-p prompt] [-U user] [-u user] [command [arg ...]]
usage: sudo [-ABbEHkNnPS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] [VAR=value]
            [-i | -s] [command [arg ...]]
usage: sudo -e [-ABkNnS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] file ...
sudo: unrecognized option '--basedir'
usage: sudo -h | -K | -k | -V
usage: sudo -v [-ABkNnS] [-g group] [-h host] [-p prompt] [-u user]
usage: sudo -l [-ABkNnS] [-g group] [-h host] [-p prompt] [-U user] [-u user] [command [arg ...]]
usage: sudo [-ABbEHkNnPS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] [VAR=value]
            [-i | -s] [command [arg ...]]
usage: sudo -e [-ABkNnS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] file ...
sudo: unrecognized option '--basedir'
usage: sudo -h | -K | -k | -V
usage: sudo -v [-ABkNnS] [-g group] [-h host] [-p prompt] [-u user]
usage: sudo -l [-ABkNnS] [-g group] [-h host] [-p prompt] [-U user] [-u user] [command [arg ...]]
usage: sudo [-ABbEHkNnPS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] [VAR=value]
            [-i | -s] [command [arg ...]]
usage: sudo -e [-ABkNnS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] file ...
sudo: unrecognized option '--basedir'
usage: sudo -h | -K | -k | -V
usage: sudo -v [-ABkNnS] [-g group] [-h host] [-p prompt] [-u user]
usage: sudo -l [-ABkNnS] [-g group] [-h host] [-p prompt] [-U user] [-u user] [command [arg ...]]
usage: sudo [-ABbEHkNnPS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] [VAR=value]
            [-i | -s] [command [arg ...]]
usage: sudo -e [-ABkNnS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] file ...
sudo: unrecognized option '--basedir'
usage: sudo -h | -K | -k | -V
usage: sudo -v [-ABkNnS] [-g group] [-h host] [-p prompt] [-u user]
usage: sudo -l [-ABkNnS] [-g group] [-h host] [-p prompt] [-U user] [-u user] [command [arg ...]]
usage: sudo [-ABbEHkNnPS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] [VAR=value]
            [-i | -s] [command [arg ...]]
usage: sudo -e [-ABkNnS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] file ...
sudo: unrecognized option '--basedir'
usage: sudo -h | -K | -k | -V
usage: sudo -v [-ABkNnS] [-g group] [-h host] [-p prompt] [-u user]
usage: sudo -l [-ABkNnS] [-g group] [-h host] [-p prompt] [-U user] [-u user] [command [arg ...]]
usage: sudo [-ABbEHkNnPS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] [VAR=value]
            [-i | -s] [command [arg ...]]
usage: sudo -e [-ABkNnS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] file ...
sudo: unrecognized option '--basedir'
usage: sudo -h | -K | -k | -V
usage: sudo -v [-ABkNnS] [-g group] [-h host] [-p prompt] [-u user]
usage: sudo -l [-ABkNnS] [-g group] [-h host] [-p prompt] [-U user] [-u user] [command [arg ...]]
usage: sudo [-ABbEHkNnPS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] [VAR=value]
            [-i | -s] [command [arg ...]]
usage: sudo -e [-ABkNnS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] file ...
sudo: unrecognized option '--basedir'
usage: sudo -h | -K | -k | -V
usage: sudo -v [-ABkNnS] [-g group] [-h host] [-p prompt] [-u user]
usage: sudo -l [-ABkNnS] [-g group] [-h host] [-p prompt] [-U user] [-u user] [command [arg ...]]
usage: sudo [-ABbEHkNnPS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] [VAR=value]
            [-i | -s] [command [arg ...]]
usage: sudo -e [-ABkNnS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] file ...
sudo: unrecognized option '--basedir'
usage: sudo -h | -K | -k | -V
usage: sudo -v [-ABkNnS] [-g group] [-h host] [-p prompt] [-u user]
usage: sudo -l [-ABkNnS] [-g group] [-h host] [-p prompt] [-U user] [-u user] [command [arg ...]]
usage: sudo [-ABbEHkNnPS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] [VAR=value]
            [-i | -s] [command [arg ...]]
usage: sudo -e [-ABkNnS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] file ...
sudo: unrecognized option '--basedir'
usage: sudo -h | -K | -k | -V
usage: sudo -v [-ABkNnS] [-g group] [-h host] [-p prompt] [-u user]
usage: sudo -l [-ABkNnS] [-g group] [-h host] [-p prompt] [-U user] [-u user] [command [arg ...]]
usage: sudo [-ABbEHkNnPS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] [VAR=value]
            [-i | -s] [command [arg ...]]
usage: sudo -e [-ABkNnS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] file ...
sudo: unrecognized option '--basedir'
usage: sudo -h | -K | -k | -V
usage: sudo -v [-ABkNnS] [-g group] [-h host] [-p prompt] [-u user]
usage: sudo -l [-ABkNnS] [-g group] [-h host] [-p prompt] [-U user] [-u user] [command [arg ...]]
usage: sudo [-ABbEHkNnPS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] [VAR=value]
            [-i | -s] [command [arg ...]]
usage: sudo -e [-ABkNnS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] file ...
Created symlink /etc/systemd/system/multi-user.target.wants/biqu-b1.service → /etc/systemd/system/biqu-b1.service.

*************************
octoprint_deploy 1.0.4
*************************

1) Add instance	    3) Add Camera	5) Utilities	    7) Update
2) Delete instance  4) Delete Camera	6) Backup Menu	    8) Quit
Select operation: 

``` 

The --basedir option is being used in 
cameras.sh
instance.sh 
prepare.sh
util.sh

Not sure what sudo-option correctly replaces --basedir, -D and -R might work.

Thank you and best regards,
SK

matthieu-sofa, 2023-09-15:
Hi Paul,
I face the same issue. I try the remove command multiple time and got the same message each time.

I can't have my instance to run (only the cam).

FYI I previously had an existing octoprint install but selected to install a most recent Octoprint.

`
*******************************************
Service - Status:
------------------
ArtillerySWX2 -      failed
cam_ArtillerySWX2 -      active
*******************************************
Only instances and cameras made with octoprint_deploy are shown

*************************
octoprint_deploy 1.0.4
*************************

1) Add instance	    3) Add Camera	5) Utilities	    7) Update
2) Delete instance  4) Delete Camera	6) Backup Menu	    8) Quit
Select operation: 8
matt@debian:~$ 
matt@debian:~$ 
matt@debian:~$ sudo octoprint_deploy/octoprint_deploy.sh remove
Remove everything? [y/n]: y
Removed /etc/systemd/system/multi-user.target.wants/ArtillerySWX2.service.

Removed /etc/systemd/system/multi-user.target.wants/cam_ArtillerySWX2.service.
Failed to stop cam_ArtillerySWX2.service: Unit cam_ArtillerySWX2.service not loaded.
Failed to disable unit: Unit file cam_ArtillerySWX2.service does not exist.
Removing system stuff

*************************
octoprint_deploy 1.0.4
*************************

1) Prepare system
2) Update
3) Quit
Select operation: 1) Prepare system
2) Update
3) Quit
Select operation: 1
Searching home directory for existing OctoPrint venv/binary.....
OctoPrint binary found at /home/matt/.local/bin/octoprint
1) Use existing binary
2) Install most recent OctoPrint
3) More information
Select option number: 2
Looking for existing OctoPrint systemd files.....

1) Ubuntu 20+, Mint, Debian, Raspberry Pi OS  3) ArchLinux				    5) Quit
2) Fedora/CentOS			      4) openSUSE
Installation type: 1
Ready to begin? [y/n]: y
Adding current user to dialout and video groups.

You have the option of setting up haproxy.
This binds instances to a name on port 80 instead of having to type the port.

Use haproxy? [y/n]: 
  invalid input 
Use haproxy? [y/n]: y
Synchronizing state of haproxy.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable haproxy
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = "en_US:en",
[... 148 lines cut ...]
            [VAR=value] [-i|-s] [<command>]
usage: sudo -e [-AknS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] file ...
sudo: unrecognized option '--basedir'
usage: sudo -h | -K | -k | -V
usage: sudo -v [-AknS] [-g group] [-h host] [-p prompt] [-u user]
usage: sudo -l [-AknS] [-g group] [-h host] [-p prompt] [-U user] [-u user] [command]
usage: sudo [-AbEHknPS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user]
            [VAR=value] [-i|-s] [<command>]
usage: sudo -e [-AknS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] file ...
sudo: unrecognized option '--basedir'
usage: sudo -h | -K | -k | -V
usage: sudo -v [-AknS] [-g group] [-h host] [-p prompt] [-u user]
usage: sudo -l [-AknS] [-g group] [-h host] [-p prompt] [-U user] [-u user] [command]
usage: sudo [-AbEHknPS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user]
            [VAR=value] [-i|-s] [<command>]
usage: sudo -e [-AknS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] file ...
sudo: unrecognized option '--basedir'
usage: sudo -h | -K | -k | -V
usage: sudo -v [-AknS] [-g group] [-h host] [-p prompt] [-u user]
usage: sudo -l [-AknS] [-g group] [-h host] [-p prompt] [-U user] [-u user] [command]
usage: sudo [-AbEHknPS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user]
            [VAR=value] [-i|-s] [<command>]
usage: sudo -e [-AknS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T timeout] [-u user] file ...
Instance must be restarted for settings to take effect. Restart now? [y/n]: 
`

paukstelis, 2023-09-15:
In that case it is not even trying to install OctoPrint. I do not know why, unless you have not provided some of the output.
Before the haproxy option it should go through the whole OctoPrint installation.

matthieu-sofa, 2023-09-18:
Removing the Octoprint in /home/mylogin/.local/bin/octoprint solved all my issues. I was able to do a fresh clean install via your script @paukstelis . My two printers are now well connected and work like a charm. Thanks @paukstelis

closed 2023-12-20 by paukstelis