paukstelis/octoprint_deploy #123: Error on 1st instance when powering up 2nd

smithlevenson, 2023-10-02:
I was in the middle of a print when I powered on my 2nd device. It created an error with both instances and the print on the 1st failed. I am using Klipper Firmware on two Ender 3 Pro v4.2.7.

paukstelis, 2023-10-02:
The only reason that would likely happen is if you connect to the wrong printer through OctoPrint. Since you are using Klipper this is not something that octoprint_deploy can know or do anything about.

closed 2023-10-02 by paukstelis

smithlevenson, 2023-10-03:
> The only reason that would likely happen is if you connect to the wrong printer through OctoPrint. Since you are using Klipper this is not something that octoprint_deploy can know or do anything about.

Thanks for the info. Just curious, is running Klipper on two instances of Octoprint too much for a pi 4?