paukstelis/octoprint_deploy #124: Unable to add a webcam to Octoprint

k5jat, 2023-10-10:
I get to the following point and it just freezes:

```
Plug your camera in via USB now (detection time-out in 1 min)
/dev/v4l/by-id/usb-046d_C922_Pro_Stream_Webcam_0A62969F-video-index0
Camera was only detected as /dev/v4l/by-id entry.
This will be used as the camera device identifier
Camera detected with serial number:
```

This is on a Le Potato running Ubuntu 22.04. The webcam, in this case, a Logitech C922 Pro, is detected but it never gets past this point. I also tried it on a different, more generic knock-off and it did the same thing.

paukstelis, 2023-10-10:
What do you mean by "it never gets past this point"? What happens next?

paukstelis, 2023-10-11:
I have pushed a fix. update octoprint_deploy and try again.

Also, I am curious what output dmesg is getting from your camera. If you can, in a terminal do the following:
`sudo dmesg -C`
plug camera in by usb
`sudo dmesg`
provide output.

k5jat, 2023-10-11:
The Le Potato died on me. Boots up but no video and won't connect to the network. I'll see if maybe flashing it with Raspbian will bring it back, though I doubt it. $35 down the tubes. If it does come back up, I do the testing you asked me to.

Have a great day!

closed 2023-12-20 by paukstelis