paukstelis/octoprint_deploy #129: Little "pacman --needed" fix

Unicod3r, 2023-10-29:
On most Arch Linux systems base-devel packages are already installed and it's a pain to redownload all of them with slow internet connection.

So adding --needed argument prevents pacman from reinstalling packages that are already installed.

KeithCu, 2023-12-29:
I just did a pull request for this:
https://github.com/paukstelis/octoprint_deploy/pull/137

This looks like a great tool, even for single instance installs. Thank you!

git4rker, 2023-12-29:
Heh, I use Mainsail now but I'm still happy that someone payed attention to my issue.

I guess I will close it, as the PR is already there.

closed 2023-12-29 by git4rker

KeithCu, 2023-12-29:
I've not installed OctoPrint yet, but I discovered this great tool, that should probably be an official part of the OctoPrint project, for those who don't use the pre-configured images.