paukstelis/octoprint_deploy #130: Camera service masked

KaeShoru, 2023-10-31:
When I installed the camera everything went well up to a point. After that, the service did not work for me and the camera did not start. Can I manually create this service?
```
Camera was only detected as /dev/v4l/by-id entry.
This will be used as the camera device identifier
Settings can be modified after initial setup in /etc/systemd/system/cam_Ender3.service

Camera Resolution [default: 640x480]:

Selected camera resolution: 640x480
Camera Framerate (use 0 for ustreamer hardware) [default: 5]:

Selected camera framerate: 5
sed: -e expression #2, char 12: unknown option to `s'
Instance must be restarted for settings to take effect. Restart now? [y/n]: y
Failed to start cam_Ender3.service: Unit cam_Ender3.service is masked.
Failed to enable unit: Unit file /etc/systemd/system/cam_Ender3.service is masked.
```

martoliod, 2023-11-09:
ive got the same issue with my Logitech c170 USB Webcam. did u solve the problem already?

KaeShoru, 2023-11-09:
> ive got the same issue with my Logitech c170 USB Webcam. did u solve the problem already?

Nope, i have Logitech camera too but C270

paukstelis, 2023-11-10:
It is not clear why this is happening, but it happened to someone else on the Discord. It doesn't have anything to do with the type of camera, but with writing the service file.

You can try what they did, which fixed it:
`sudo systemctl unmask cam_instancename`
run octoprint_deploy, remove the camera, then reinstall the camera.

KaeShoru, 2023-11-10:
> It is not clear why this is happening, but it happened to someone else on the Discord. It doesn't have anything to do with the type of camera, but with writing the service file.
>
> You can try what [...]

I've tried this many times and it doesn't help. Moreover, the service itself is empty for some reason (and therefore the unmask command deletes it). Is it possible to find out the approximate content of this service so that I can substitute it with my own adjustments? I also want to note that when I start searching for USB ports, the script does not show anything. Well, in addition, I want to note that I’m running all this on an Orange Pi Zero 2

paukstelis, 2023-11-10:
I suspect it is masked because there is invalid information being supplied to the script so the service file fails.

Try installing the camera with all default values (don't change resolution or frame rate).

If it searches for cameras and doesn't show anything that means your camera is not detected as a v4l camera and won't work directly.

you can verify this:
unplug camera
`sudo dmesg -C`
plug camera in
`sudo dmesg`
provide output

KaeShoru, 2023-11-10:
> I suspect it is masked because there is invalid information being supplied to the script so the service file fails.
>
> Try installing the camera with all default values (don't change resolution or [...]

The first time I installed Octoprint on this device with this script, the camera was detected correctly and worked perfectly (camera works in octoprint). But I had to redo everything a little due to the fact that Orange Pi Zero has only 1 usb port. The first time, the camera and printer were connected through the hub, but the printer kept disconnecting. Therefore, I connected the printer via gpio usb, and the camera via a regular one. It was after this that the camera stopped working.

sftwninja, 2023-11-27:
I did some snooping and found that `sed` was getting confused by some double slashes appearing in one of the `-e` options. (Looks like it was choking on `CAMPORT`)

`sed`'s pretty flexible, so swapping out `/` with `@` as a delimiter worked fine:
```
-        sed -e "s/OCTOUSER/$OCTOUSER/" \
-        -e "s/OCTOCAM/$CAMDEVICE/" \
-        -e "s/RESOLUTION/$RESOLUTION/" \
-        -e "s/FRAMERATE/$FRAMERATE/" \
-        -e "s/CAMPORT/$CAMPORT/" > $SCRIPTDIR/cam${INUM}_$INSTANCE.service
+        sed -e "s@OCTOUSER@$OCTOUSER@" \
+        -e "s@OCTOCAM@$CAMDEVICE@" \
+        -e "s@RESOLUTION@$RESOLUTION@" \
+        -e "s@FRAMERATE@$FRAMERATE@" \
+        -e "s@CAMPORT@$CAMPORT@" > $SCRIPTDIR/cam${INUM}_$INSTANCE.service```

However, and I've not looked into it to any degree, the `CAMPORT` is getting set from `/dev/OCTOCAM` to `/dev//dev/the-actual-cam`.

As a note, I'm using the latest 1.0.5 changes that fix `by-id`.

paukstelis, 2023-11-28:
> I did some snooping and found that `sed` was getting confused by some double slashes appearing in one of the `-e` options. (Looks like it was choking on `CAMPORT`)
> 
> `sed`'s pretty flexible, so swapping out `/` with `@` as a delimiter worked fine:
> 
> ```
> @@ -57,11 +57,11 @@ write_camera() {
>      #mjpg-streamer
>      if [ "$STREAMER" == mjpg-streamer ]; then
>          cat $SCRIPTDIR/octocam_mjpg.service | \
> -        sed -e "s/OCTOUSER/$OCTOUSER/" \
> -        -e "s/OCTOCAM/$CAMDEVICE/" \
> -        -e "s/RESOLUTION/$RESOLUTION/" \
> -        -e "s/FRAMERATE/$FRAMERATE/" \
> -        -e "s/CAMPORT/$CAMPORT/" > $SCRIPTDIR/cam${INUM}_$INSTANCE.service
> +        sed -e "s@OCTOUSER@$OCTOUSER@" \
> +        -e "s@OCTOCAM@$CAMDEVICE@" \
> +        -e "s@RESOLUTION@$RESOLUTION@" \
> +        -e "s@FRAMERATE@$FRAMERATE@" \
> +        -e "s@CAMPORT@$CAMPORT@" > $SCRIPTDIR/cam${INUM}_$INSTANCE.service
> ```
> 
> However, and I've not looked into it to any degree, the `CAMPORT` is getting set from `/dev/OCTOCAM` to `/dev//dev/the-actual-cam`.
> 
> As a note, I'm using the latest 1.0.5 changes that fix `by-id`.

I have to say, this doesn't really make much sense. Not that anything is wrong, just that octoprint_deploy doesn't use mjpg_streamer at all anymore. Make changes in the mjpg-streamer section there should have absolutely no impact on it writing out the service file.

I am curious what OS this is on, because I haven't seen this with any of the versions of sed that I have used.

martoliod, 2023-12-02:
> I suspect it is masked because there is invalid information being supplied to the script so the service file fails.
>
> Try installing the camera with all default values (don't change resolution or [...]

i get the following output from `sudo dmesg`:

```
[ 1532.896937] usb 5-1: new high-speed USB device number 4 using xhci-hcd
[ 1533.112431] usb 5-1: New USB device found, idVendor=046d, idProduct=082b, bcdDevice=28.c5
[ 1533.112464] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1533.112484] usb 5-1: Product: Webcam C170
[ 1533.112502] usb 5-1: Manufacturer:
[ 1533.130234] uvcvideo: Found UVC 1.00 device Webcam C170 (046d:082b)
```

martoliod, 2023-12-02:
> I did some snooping and found that `sed` was getting confused by some double slashes appearing in one of the `-e` options. (Looks like it was choking on `CAMPORT`)
>
> `sed`'s pretty flexible, so [...]

does this solve the problem for you? Unfortunately, I still get the same error after applying these changes....

paukstelis, 2023-12-02:
Can you join the octoprint_deploy Discord (invite link is the README)? It is easier to help diagnose issues there.

KaeShoru, 2023-12-02:
I ended up filling the service manually as it was done in the substitution file. Now everything works

Blisk, 2023-12-14:
I have the same problem, the camera not working and still don't know what to do?
 sudo dmesg
[255379.687808] usb 1-1.2: new high-speed USB device number 15 using dwc_otg
[255382.129302] usb 1-1.2: New USB device found, idVendor=046d, idProduct=085b, bcdDevice= 0.16
[255382.129342] usb 1-1.2: New USB device strings: Mfr=0, Product=2, SerialNumber=1
[255382.129358] usb 1-1.2: Product: Logitech Webcam C925e
[255382.129371] usb 1-1.2: SerialNumber: 5516DFCF
[255382.130356] usb 1-1.2: Found UVC 1.00 device Logitech Webcam C925e (046d:085b)
[255382.134683] input: Logitech Webcam C925e as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/input/input8

paukstelis, 2023-12-14:
A camera can not work for lots of reasons. If you have an issue with a masked service file, provide the contents of octoprint_deploy/cam_instancename.attempt

Blisk, 2023-12-14:
[Unit]
Description=the OctoPi(buntu) ustreamer daemon with the user specified config
After=network.online.target
Wants=network.online.target

[Service]
User=damjan
ExecStart=/home/damjan/ustreamer/ustreamer -d /dev/cam_ENDER3 -s 0.0.0.0 -m MJPEG -r 640x480 -f 5 -p 8002 --device-timeout 8 --device-error-delay 8

[Install]
WantedBy=multi-user.target

paukstelis, 2023-12-14:
There is nothing wrong with that. 
Re-run octoprint_deploy (sudo octoprint_deploy/octoprint_deploy.sh) and choose diagnostic info from the Utilities menu.

Blisk, 2023-12-14:
Ok here it is. For info I have two cameras for two differend printers. none of cam works. Before I add second printer, camera works ok. Both cameras are the same.
___________________________________________________

octoprint_deploy diagnostic information. Please provide ALL output for support help
**************************************
/etc/octoprint_deploy
**************************************
octoexec: /home/damjan/oprint/bin/octoprint
octopip: /home/damjan/oprint/bin/pip
haproxy: true
octopi: true
**************************************
/etc/octoprint_instances
**************************************
instance:octoprint port:5000 udev:false
instance:ENDER3 port:5001 udev:true
**************************************
/etc/octoprint_cameras
**************************************
camera:cam_octoprint port:8001 udev:true
camera:cam_ENDER3 port:8002 udev:true
**************************************
/etc/udev/rules.d/99-octoprint.rules
**************************************
SUBSYSTEM=="tty", ATTRS{serial}=="20873155574B", SYMLINK+="octo_ENDER3"
SUBSYSTEM=="video4linux", ATTRS{serial}=="5516DFCF", ATTR{index}=="0", SYMLINK+="cam_octoprint"
SUBSYSTEM=="video4linux", ATTRS{serial}=="0D1A8FCF", ATTR{index}=="0", SYMLINK+="cam_ENDER3"
lrwxrwxrwx 1 root root 7 Dec 14 12:51 /dev/octo_ENDER3 -> ttyACM0
lrwxrwxrwx 1 root root 6 Dec 14 12:47 /dev/cam_ENDER3 -> video2
lrwxrwxrwx 1 root root 6 Dec 14 12:47 /dev/cam_octoprint -> video0
**************************************
● octoprint.service - The snappy web interface for your 3D printer
     Loaded: loaded (/etc/systemd/system/octoprint.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2023-12-14 12:29:51 CET; 6h ago
   Main PID: 389 (octoprint)
      Tasks: 47 (limit: 1472)
        CPU: 6min 59.799s
     CGroup: /system.slice/octoprint.service
             └─389 /home/damjan/oprint/bin/python /home/damjan/oprint/bin/octoprint serve --host=127.0.0.1 --port=5000

Dec 14 17:01:57 pi octoprint[389]: 2023-12-14 17:01:57,620 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
Dec 14 17:01:57 pi octoprint[389]: 2023-12-14 17:01:57,750 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
Dec 14 17:01:59 pi octoprint[389]: 2023-12-14 17:01:59,660 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
Dec 14 17:02:01 pi octoprint[389]: 2023-12-14 17:02:01,836 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
Dec 14 17:15:27 pi octoprint[389]: 2023-12-14 17:15:27,808 - octoprint.server.heartbeat - INFO - Server heartbeat <3
Dec 14 17:30:27 pi octoprint[389]: 2023-12-14 17:30:27,809 - octoprint.server.heartbeat - INFO - Server heartbeat <3
Dec 14 17:45:27 pi octoprint[389]: 2023-12-14 17:45:27,810 - octoprint.server.heartbeat - INFO - Server heartbeat <3
Dec 14 18:00:27 pi octoprint[389]: 2023-12-14 18:00:27,812 - octoprint.server.heartbeat - INFO - Server heartbeat <3
Dec 14 18:15:27 pi octoprint[389]: 2023-12-14 18:15:27,813 - octoprint.server.heartbeat - INFO - Server heartbeat <3
Dec 14 18:30:27 pi octoprint[389]: 2023-12-14 18:30:27,815 - octoprint.server.heartbeat - INFO - Server heartbeat <3
None
None
{'bitrate': '10000k',
 'cleanTmpAfterDays': 7,
 'defaultWebcam': 'classic',
 'ffmpeg': None,
 'ffmpegCommandline': '{ffmpeg} -framerate {fps} -i "{input}" -vcodec '
                      '{videocodec} -threads {threads} -b:v {bitrate} -f '
                      '{containerformat} -y {filters} "{output}"',
 'ffmpegThreads': 1,
 'ffmpegThumbnailCommandline': '{ffmpeg} -sseof -1 -i "{input}" -update 1 -q:v '
                               '0.7 "{output}"',
 'ffmpegVideoCodec': 'libx264',
 'snapshotWebcam': 'classic',
 'timelapse': {'fps': 25,
               'options': {'capturePostRoll': None,
                           'interval': None,
                           'retractionZHop': None},
               'postRoll': 0,
               'type': 'off'},
[... 68 lines cut ...]
Dec 14 12:47:54 pi systemd[1]: cam_ENDER3.service: Main process exited, code=exited, status=203/EXEC
Dec 14 12:47:54 pi systemd[1]: cam_ENDER3.service: Failed with result 'exit-code'.
**************************************
● haproxy.service - HAProxy Load Balancer
     Loaded: loaded (/lib/systemd/system/haproxy.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2023-12-14 12:47:54 CET; 5h 49min ago
       Docs: man:haproxy(1)
             file:/usr/share/doc/haproxy/configuration.txt.gz
   Main PID: 1469 (haproxy)
      Tasks: 5 (limit: 1472)
        CPU: 7.402s
     CGroup: /system.slice/haproxy.service
             ├─1469 /usr/sbin/haproxy -Ws -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid -S /run/haproxy-master.sock
             └─1471 /usr/sbin/haproxy -Ws -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid -S /run/haproxy-master.sock

Dec 14 17:13:46 pi haproxy[1471]: ::ffff:192.168.0.130:43256 [14/Dec/2023:17:13:46.536] public octoprint/octoprint1 0/0/1/50/51 200 931 - - ---- 2/2/1/1/0 0/0 "GET /plugin/discovery/discovery.xml HTTP/1.1"
Dec 14 17:13:46 pi haproxy[1471]: ::ffff:192.168.0.130:43257 [14/Dec/2023:17:13:46.537] public octoprint/octoprint1 0/0/0/92/92 200 931 - - ---- 2/2/0/0/0 0/0 "GET /plugin/discovery/discovery.xml HTTP/1.1"
[... cut ...]
Dec 14 17:15:17 pi haproxy[1471]: ::ffff:192.168.0.130:43257 [14/Dec/2023:17:15:17.425] public octoprint/octoprint1 0/0/0/88/88 200 931 - - ---- 2/2/0/0/0 0/0 "GET /plugin/discovery/discovery.xml HTTP/1.1"
Dec 14 17:15:50 pi haproxy[1471]: ::ffff:192.168.0.130:43257 [14/Dec/2023:17:15:50.026] public octoprint/octoprint1 0/0/0/46/46 200 931 - - ---- 2/2/1/1/0 0/0 "GET /plugin/discovery/discovery.xml HTTP/1.1"
Dec 14 17:15:50 pi haproxy[1471]: ::ffff:192.168.0.130:43256 [14/Dec/2023:17:15:50.029] public octoprint/octoprint1 0/0/0/84/84 200 931 - - ---- 2/2/0/0/0 0/0 "GET /plugin/discovery/discovery.xml HTTP/1.1"

paukstelis, 2023-12-14:
Your issue is that ustreamer has not been installed. Choose 'Change Streamer' in Utilities menu. Remove all cameras, and reinstall them.

Blisk, 2023-12-16:
> Your issue is that ustreamer has not been installed. Choose 'Change Streamer' in Utilities menu. Remove all cameras, and reinstall them.

No still not working, it is the same. I removed both cameras and add it again and nothing. After I install ustreamer

Blisk, 2023-12-16:
> Your issue is that ustreamer has not been installed. Choose 'Change Streamer' in Utilities menu. Remove all cameras, and reinstall them.

I did this second time and now it works, both cameras are on.
Thank you