paukstelis/octoprint_deploy #131: Pi5, Bookworm, Install Failure

addohm, 2023-11-14:
I know its early in the Pi5 release but thought you'd want to know that the install script fails to create and\or query anything in `/etc/octoprint_instances`.

```
*************************
octoprint_deploy 1.0.5
*************************

1) Add instance     3) Add Camera       5) Utilities        7) Update
2) Delete instance  4) Delete Camera    6) Backup Menu      8) Quit
Select operation: 3
Adding camera
cat: /etc/octoprint_instances: No such file or directory
1) Quit
Select instance number to add camera to: 1
```

addohm, 2023-11-14:
Theres also this.  All my backups are zip files.  Its got the correct location but needs to include zip files.

```
1) Create Backup
2) Restore Backup
3) Quit
Select an option: 2

ls: cannot access '/home/addohm/*.tar.gz': No such file or directory
1) Quit
```

paukstelis, 2023-11-14:
1. You have to add an instance before adding cameras. You haven't done that yet.
2. OctoPrint and octoprint_deploy backups are totally different and aren't cross-compatible.

closed 2023-11-14 by paukstelis