paukstelis/octoprint_deploy #136: Cam Install failure

BustedTaint, 2023-12-29:
Trying to get my 2nd cam for my second printer installed, and it keeps failing with the following. It looks like it doesn't create the service?
-----------------Start-----------------------------
Verify the camera is currently unplugged from USB.....
Is the camera you are trying to detect unplugged from USB? [y/n]: y
Plug your camera in via USB now (detection time-out in 1 min)
/dev/v4l/by-id/usb-046d_HD_Webcam_C525_56E13B20-video-index0
Camera detected with serial number: 56E13B20
No duplicate serial number detected.
Settings can be modified after initial setup in /etc/systemd/system/cam_X3.service

Camera Resolution [default: 640x480]:

Selected camera resolution: 640x480
Camera Framerate (use 0 for ustreamer hardware) [default: 5]:
0
Selected camera framerate: 0
Ready to write all changes. Do you want to proceed? [y/n]: y
Uniquifying instance...
cp: cannot stat '/home/xxxx/octoprint_deploy/cam_X3.service': No such file or directory
mv: cannot stat '/home/xxxx/octoprint_deploy/cam_X3.service': No such file or directory
Instance must be restarted for settings to take effect. Restart now? [y/n]: y
Created symlink /etc/systemd/system/multi-user.target.wants/X3.service → /etc/systemd/system/X3.service.
Failed to start cam_X3.service: Unit cam_X3.service not found.
Failed to enable unit: Unit file cam_X3.service does not exist.
-----------------End-----------------------------

paukstelis, 2023-12-29:
I'm seeing this reported more often and I still don't know what the issue is.
Re-run octoprint_deploy and use the diagnostic output from the utility menu. Provide all output.

BustedTaint, 2023-12-30:
So I've done about 5 complete fresh builds. RPi 3B+, flashed fresh OctoPi with RPi Imager, loaded up first instance, all running smooth before octoprint_deploy, same results. Was attempting to perform another fresh build to get the diagnostic output, but this time it's not starting the second instance.
One thing that may be notable, when running "Prepare System", it doesn't run/install ustreamer. 
"sudo octoprint_deploy/octoprint_deploy.sh"
- OctoPi instance created
- Udev runs for the current instance (printer and camera)
- Never prompted for ustreamer
......and then add second printer/instance, seems fine. Try and add camera, it then notes that ustreamer is not being used. Go into utilities, switch to ustreamer, and then try and install the camera is when those errors would occur. 
If I get a chance, I will try and wipe and try again, but for today, I'm just going to wipe, start fresh and manually create another instance.
@paukstelis thanks for the awesome work on this!  This could just be a hardware issue on my end, as this pi has been running prints for years.

paukstelis, 2024-01-06:
This should now be fixed. I believe the error was in some duplication of the streamer setting in /etc/octoprint_deploy. Removing a cameras, switching streamer to ustreamer, and reinstalling should work.

closed 2024-01-06 by paukstelis

BustedTaint, 2024-01-08:
@paukstelis good to hear, if I rebuild, which at some point will happen, I will try it. For now, I did use the scripts to build out additional instances, and just manually installed cams in the latest cam-streamer. ......again, thank you for all of the work  that saves lots of manual time and frustration, an awesome contribution to the community!