paukstelis/octoprint_deploy #139: gentoo and generic GNU/linux support

petaflot, 2023-12-30:
it's all in the title, that would be nice.

eventually, when packages need to be installed, display a message with the required packages to let the user do this by hand ; since some distros do not use system, prompt the user to restart services by hand if required

paukstelis, 2024-01-06:
I have no plans to add gentoo. I've never used it and it seems like most people that have the patience to use it should have enough linux-fu to run some commands to install OctoPrint.

closed 2024-01-06 by paukstelis

petaflot, 2024-01-06:
I don't quite agree with you. And I'm gonna be harsh.

I thought - and still want to believe - that by working on open software we were all seeking to create an entire eco-system where minds and spirits could be free.

Unfortunately it seems to me you're just using it for your very own purpose. I do have some "fu", but right now I have more important matters to deal with. Not that I don't want to, I just don't have the energy. 

> I've never used it

well maybe you should give it a try. And trust me, it was way more complicated a few years back ; something like when configuring X required finding the right combination of refresh rates and frequencies both graphics cards and monitor would support.

>> side note: native 64-bits was available on the Pi with Gentoo at least 4 years before it was on any deb-based system

hmm I wonder I much done by the "gentooists" profited to those who make the distribution you're running. It seems to me you're confusing "Linux" with "GNU/Linux".

At the very least, I would have hoped you would leave this issue open ; you (@paukstelis) are probably not the only one who might want to give it a try. At the very least, I was asking for an option to prompt the user to manually install deps and run whatever commands (in a generic form, such as displaying commands required for each supported OS in order to help figure out what to actually do) are required on their OS (there are others like Slackware, SMGL, LFS...), in the right order, so *maybe* they could contribute and this would make it easier for them to do so by not requiring them to read hundreds of lines of code instead of just walking away to something else. Because trust me, a fair amount of gentooists do read so much code every day that you would get a nausea from (and I write this because you just bluntly closed the issue with one of the crappiest argument I have ever heard from a dev).

So at the very least, re-open this issue and let someone - could be me for all I know - try to address it.

paukstelis, 2024-01-06:
Yes, I made this for my own purposes and it will continue to be that way. If you want something different, create a PR, or fork it and do your own thing. I owe nothing to you, gentooists, or anyone else as far as this project is concerned.

petaflot, 2024-01-06:
thank you for giving me the best reason to never use octoprint, ever.

petaflot, 2024-01-06:
like... this feature request could have been a reminder I set myself - where it actually belongs - to improve octoprint. yet you discarded it just because you don't bother trying out gentoo (amongst other distros which I've named).

so what you're suggesting is I should make a fork and set a bug report on my own fork - which probably no-one will ever see unless they're masochists - therefore preventing others from thinking "well yeah I could have the time for that" or "uh what Gentoo/SMGL/LFS? never heard of it, I'm going to give it a look" and actually addressing the issue while I'm busy with (what I consider to be) more important stuff, just because... what? you want the open issues count on octoprint to be as low as possible?

I consider this to be very lame, and honestly I don't give a crap if you don't like my answer. I'm done with you and your project.

paukstelis, 2024-01-06:
1. I am not affiliated with OctoPrint in any way. This is just a tool for installing OctoPrint.
2. It is removed as an open issue because I am not going to work on it. 
3. If you really want Gentoo added, you or someone else are welcome to do the work.

Good bye.