paukstelis/octoprint_deploy #141: [Enhancement] add camera-streamer option

b-morgan, 2024-01-06:
Please add an option (and the code) to install and configure [camera-streamer](https://github.com/ayufan/camera-streamer).

I realize this won't be simple (as I've tried it unsuccessfully) but I believe that with the cooperation with the camera-streamer author and various OctoPrint gurus (including Gina) this would be an excellent addition to this project.

For non-Raspberry Pi systems, this may expand to installing and configuring [libcamera](https://libcamera.org/).

paukstelis, 2024-01-06:
Everything is already there to install camera-streamer except for setting up service files. There are several issues with using camera-streamer, which is why I haven't actually included it as of yet. The primary reason is that since the new camera stack for OctoPi is using it, it will lead to confusion on HOW cameras are setup. Did someone create a camera-streamer service with octoprint_deploy, or did they use OctoPi scripts to add the camera? A second reason is that the more options that are present, the harder it is help people debug issues. This is the reason I removed mjpg-streamer entirely. I'm not convinced that camera-streamer is so much better than ustreamer that it warrants making this change.

b-morgan, 2024-01-06:
Except that ustreamer doesn't support CSI cameras (i.e. the Raspberry Pi Camera 3). If there is an alternative that supports these cameras then I'd accept it.

paukstelis, 2024-01-06:
This is the issue that I considered when I chose to not include camera-streamer. If you are using a RPi 3 camera, you can always just use OctoPi and have camera-streamer immediately ready to go. From there, you can use the OctoPi tools to install any additional cameras. I would like to keep octoprint_deploy image agnostic and constantly having to make exceptions for OctoPi didn't seem worth it anymore.

b-morgan, 2024-01-06:
Except that RPi 5 support is currently missing from OctoPi 1.0.0 (either camera stack).

I have an [Orange Pi 3B](http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-3B.html) and an [Pine64 Quartz 64 B](https://www.pine64.org/quartz64b/) both of which have CSI interfaces. I used octoprint-deploy on both successfully. There are other Orange Pi offerings that also include CSI interfaces. I just don't own any of them. 

I also have a [Libre Computer Le Potato AML-S905X-CC](https://libre.computer/products/aml-s905x-cc/) but no CSI so ustreamer is the only option (and it works).

I understand your reluctance but I believe some solution for the CSI will be "image agnostic" and apply to more systems that just Raspberry Pi (and OctoPi).

paukstelis, 2024-01-06:
I would be looking for suggestions on how to actually setup CSI cameras with octoprint_deploy. Right now, I only have the `picam` command line argument, which basically just defaults to using /dev/video0 (I believe) for the next camera that is setup. It is much easier to do USB detection of cameras.