paukstelis/octoprint_deploy #142: Restoring from backup

westmorland1066, 2024-01-12:
I took backups of 2 instances through the octodeploy script (1.0.3) so I could do a fresh install of OctoPi 1.0.  On my new Octopi/deploy server I create an instance with the same name as my old instance and then try to restore the backup and it fails with the error below.   I get the same whether I try to restore from the command line or through the octoprint instance.
Filename is Ender3v2-01_2024-01-11_backup.tar.gz

```
Uploading backup, this can take a while. Please wait...
 
Restoring from backup...
 
Error while running restore
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/backup/__init__.py", line 1181, in _restore_backup
    with zipfile.ZipFile(path, "r") as zip:
  File "/usr/lib/python3.9/zipfile.py", line 1257, in __init__
    self._RealGetContents()
  File "/usr/lib/python3.9/zipfile.py", line 1324, in _RealGetContents
    raise BadZipFile("File is not a zip file")
zipfile.BadZipFile: File is not a zip file
 
Restore failed! Check the above output and octoprint.log for reasons as to why.
Removing temporary zip
```

paukstelis, 2024-01-12:
There are two separate backup systems: the native OctoPrint back, which creates a zip file, and the octoprint_deploy backup which creates a gzipped tar file. They are not compatible with one another.

Since you created your backup with octoprint_deploy, you can use octoprint_deploy to restore the backup. Place the tar.gz file in your /home/$USER directory, run octoprint_deploy and do a restore backup.

westmorland1066, 2024-01-12:
Ok, so that was dumb on my part.

So I went through octoprint_deploy  restore backup.  It unzips the files, though it complains about an unexpected End of File on one of them.  I can see all the files in the directories, however, when I do a list instances under the Utilities menu it only shows the default octoprint instance.  Neither of the restored instances are listed, and if I try to access them through the browser I get a 404 response.  Rebooting the system didn't change that.

paukstelis, 2024-01-12:
You need to recreate the instances (using the same instance names) with octoprint_deploy, then do the restore. This is described in the README file.

westmorland1066, 2024-01-12:
Thanks Paul.  I don't know how I missed that.  It's a great piece of software.

closed 2024-01-12 by westmorland1066