paukstelis/octoprint_deploy #143: No plugin install, or admin user prompt.

skewll, 2024-01-14:
Hi, I am running a fresh install of Ubuntu Server 23.1 on RPI4. 

Everything seems to be working fine, no errors. But I never get the following prompts:

- You will be prompted if you want to setup the admin user and do the first run wizard via the commandline.
- You will be prompted if you want to install recommended plugins.

I ran this for the heck of it, and nothing happened: 
```
sudo octoprint_deploy/plugins.sh
```

paukstelis, 2024-01-14:
Running that won't do anything.

What do you get when you run `Prepare System`? If you can paste the output that would be useful.

skewll, 2024-01-14:
Sorry, I ran the remove command, started over and it seems to be working as expected now. Perhaps I did something out of order the first time.

closed 2024-01-14 by skewll

skewll, 2024-01-14:
It may actually be a small bug.

When first stating, if on the first printer you do not allow auto detect, you will never ever get the admin user creation or the plugin prompt. You can add printers all day, but those prompts are never triggered.

However, if you do allow the first printer to auto connect, you get the option for admin user and plug install. Then after it says you can start installing printers.

Perhaps I misunderstood, but I thought the first printer it was possible to set up as a "template" printer, even though I realize that is no  longer necessary.

I was trying to make my first printer profile the Master template, so I could reference that "printer" when creating new printers each time. Also, my old setup started at port 5001 so I wanted to get all my printers on the same ports as before.

paukstelis, 2024-01-14:
There is no way to add an admin user if no instance has been created. It is a bit  of a chicken/egg thing. Your easiest solution is to do the auto-detection using one of your printers (probably the "last" one you want to use), add the admin user, setup the next instance with your "first" printer (this will put it on 5001). Then you can remove the udev rule for "template" instance and disable it: `sudo systemctl disable instancename; sudo systemctl stop instancename`

paukstelis, 2024-01-14:
Actually, looking through at this again (it has been a while). Just don't plug in a printer when you do the first instance install. It will continue and still make an instance.

skewll, 2024-01-15:
If I don't auto detect on the very first printer after install, I never ever get the admin/plug prompts again. Ever. All way I ever get those prompts is if auto detecting first first printer.

Also, now that I got past that. I am having trouble connecting. It connected fine before I was able to create the admin/plugging/wizard from the prompt.  But Now, I cannot connect, I get 500 api error in console log.

I have tried running the restart_all. But it shows my printer as failed. And its not running at all. So I guess I am going to start all over again and see if it something I am still doing wrong.

Has anyone ever been successful at connecting a pi-camera-v2 to one of these instances? I see it detect usb camera, whats the trick to connect to my pi camera connected to the same machine?

paukstelis, 2024-01-15:
Again, the answer in this case is to ALWAYS do auto-detect. When you run Prepare System for the first instance creation, simply don't plug in a printer and wait 1 minute. It will time-out, the instances will be created, and you will get the admin user and plugin prompt.

To use Pi cam v2, setup your instance, quit, and re-run octoprint_deploy with the picam option. `sudo octoprint_deploy/octoprint_deploy.sh picam`. The next camera you setup will default to /dev/video0.

skewll, 2024-01-15:
That worked. Everything seems to work. Thanks a million for giving your time to help me out.

I am seeing "No signal" on the pi cam. But I got it connected I guess. I will dig around to see what I need to do to make see its running correctly under ubuntu server 23.1

I made my first print Master, then each other I add the config and gcode of the Master. I keep getting this on each printer so far:

rsync: [sender] send_files failed to open "/home/pi/.Master/config.yaml": Permission denied (13)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]

Is this because I never got the option for the Wizard by doing that? Since there was no printer to be detected on the first? I am thinking the config has something to do with what happens after you launch the octoprint app and set profiles and such? So My thought is that confi.yaml doesn't exist?

Everything seems fine. I can visit the url of each printer, login with master, I get the wizard to set up my printer, name it, give anonamous usage tracke etc for each instance. All the shared plugins are there. Any idea what causes the rsync error?

paukstelis, 2024-01-15:
For the Picam, you may need to change the device in the udev rules file.
`sudo nano /etc/udev/rules.d/99-octoprint.rules`

and change the camera line where it has `camera0` to `video0`

paukstelis, 2024-01-15:
Yes, I think that the Master instance doesn't contain anything, which is why there is an error.
I would just remove everything, and include a `rm -rf ~/.Master` to make sure all that is deleted and run through the process again.

skewll, 2024-01-15:
Okay, I just tried to use a printer that I already set up this time, other then the Master. 

```
Copying template files....
Uniquifying instance...
sed: -e expression #2, char 12: unknown option to `s'
```
I copied this from a profile with a webcam.

Even though I selected no camera, after the lines above, it showed camera errors.

Seems to work though. But the camera got copied over to this printer. 

I tried removing the camera from this printer. It seemed to work. I restarted, no everything has failed to start.

```
sudo octoprint_deploy/octoprint_deploy.sh restart_all 
```

Then I check the "Instance Status" and all printer are failed. Cameras are active.  Thinking back, that is the command that made my first install inaccessible. I had to run that command after changing the shared folder. It showed Trying to restart each printer, no errors. But everything fails.

rebooting my system, everything is still failed.

skewll, 2024-01-15:
For some reason, deleting the master printer also removed all the camera with it :-(

1) Master
2) Molly
3) Holly
4) Dolly
5) Stella
6) Quit
Select instance number to remove: 1
Selected instance to remove: Master
Do you want to remove everything associated with this instance? [y/n]: y
Removed "/etc/systemd/system/multi-user.target.wants/Master.service".
Removed "/etc/systemd/system/multi-user.target.wants/cam_Molly.service".
Removed "/etc/systemd/system/multi-user.target.wants/cam_Dolly.service".

skewll, 2024-01-15:
I have undone the "Custom" shared uploads link back to sharing with one of the printer. Rebooting. Cant seem to get any of the printers active again.

Ive tried the restart command, even after rebooting. Same issue.

I will start from scratch and do some more trouble shooting.

I was close.

skewll, 2024-01-15:
It appears that the "shared uploads" is what is breaking things.

Ive tried with 4 printers, 2 printers, and just one printer. New directory, existing directory. It always breaks things at that point.

As soon as I add a shared directory. It says to restart instances. Before I get the chance to run the restart_All command, the octoprint UI reloads and I can no longer connect. In inspection is shows active. Once I run the restart_all command it goes to failed. octoprint UI shows Server Offline.

```
octoprint_deploy diagnostic information. Please provide ALL output for support help
**************************************
/etc/octoprint_deploy
**************************************
haproxy: true
streamer: ustreamer
octoexec: /home/pi/OctoPrint/bin/octoprint
octopip: /home/pi/OctoPrint/bin/pip
**************************************
/etc/octoprint_instances
**************************************
instance:Holly port:5001 udev:true
cat: /etc/octoprint_cameras: No such file or directory
**************************************
/etc/octoprint_cameras
**************************************
**************************************
/etc/udev/rules.d/99-octoprint.rules
**************************************
SUBSYSTEM=="tty", ATTRS{serial}=="CZPX2420X004XK93651", SYMLINK+="octo_Holly"
lrwxrwxrwx 1 root root 7 Jan 14 21:33 /dev/octo_Holly -> ttyACM1
lrwxrwxrwx 1 root root 7 Jan 14 21:33 /dev/octo_Molly -> ttyACM0
ls: cannot access '/dev/cam*': No such file or directory
**************************************
× Holly.service - The snappy web interface for your 3D printer
     Loaded: loaded (/etc/systemd/system/Holly.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Sun 2024-01-14 21:45:03 PST; 26s ago
   Duration: 546ms
    Process: 7251 ExecStart=/home/pi/OctoPrint/bin/octoprint serve --config=${CONFIGFILE} --basedir=${BASEDIR} --port=${PORT} (code=exited, status=2)
   Main PID: 7251 (code=exited, status=2)
        CPU: 525ms

Jan 14 21:45:03 printfarm systemd[1]: Started Holly.service - The snappy web interface for your 3D printer.
Jan 14 21:45:03 printfarm octoprint[7251]: Usage: octoprint serve [OPTIONS]
Jan 14 21:45:03 printfarm octoprint[7251]: Try 'octoprint serve --help' for help.
Jan 14 21:45:03 printfarm octoprint[7251]: Error: Invalid value for '--config' / '-c': Path '/home/pi/.Holly/config.yaml' is not readable.
Jan 14 21:45:03 printfarm systemd[1]: Holly.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Jan 14 21:45:03 printfarm systemd[1]: Holly.service: Failed with result 'exit-code'.
None
None
{'bitrate': '10000k',
 'cleanTmpAfterDays': 7,
 'defaultWebcam': 'classic',
 'ffmpeg': None,
 'ffmpegCommandline': '{ffmpeg} -framerate {fps} -i "{input}" -vcodec '
                      '{videocodec} -threads {threads} -b:v {bitrate} -f '
                      '{containerformat} -y {filters} "{output}"',
 'ffmpegThreads': 1,
 'ffmpegThumbnailCommandline': '{ffmpeg} -sseof -1 -i "{input}" -update 1 -q:v '
                               '0.7 "{output}"',
 'ffmpegVideoCodec': 'libx264',
 'snapshotWebcam': 'classic',
 'timelapse': {'fps': 25,
               'options': {'capturePostRoll': None,
                           'interval': None,
                           'retractionZHop': None},
               'postRoll': 0,
               'type': 'off'},
 'timelapseEnabled': True,
 'watermark': True,
 'webcamEnabled': True}
cat: /etc/octoprint_cameras: No such file or directory
**************************************
● haproxy.service - HAProxy Load Balancer
     Loaded: loaded (/lib/systemd/system/haproxy.service; enabled; preset: enabled)
     Active: active (running) since Sun 2024-01-14 21:37:53 PST; 7min ago
       Docs: man:haproxy(1)
             file:/usr/share/doc/haproxy/configuration.txt.gz
   Main PID: 6227 (haproxy)
      Tasks: 5 (limit: 4430)
     Memory: 7.1M
        CPU: 697ms
     CGroup: /system.slice/haproxy.service
             ├─6227 /usr/sbin/haproxy -Ws -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid -S /run/haproxy-master.sock
             └─6229 /usr/sbin/haproxy -Ws -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid -S /run/haproxy-master.sock

Jan 14 21:37:53 printfarm systemd[1]: Starting haproxy.service - HAProxy Load Balancer...
Jan 14 21:37:53 printfarm haproxy[6227]: [NOTICE]   (6227) : New worker (6229) forked
Jan 14 21:37:53 printfarm haproxy[6227]: [NOTICE]   (6227) : Loading success.
Jan 14 21:37:53 printfarm systemd[1]: Started haproxy.service - HAProxy Load Balancer.
```

paukstelis, 2024-01-15:
Ok. I think I know the problem. That function is executing the command as root, so it is changing the permissions on the config.yaml.

You can try fixing it directly: `sudo chown pi ~/.Holly/config.yaml` then restart the instance and see if it starts correctly.

paukstelis, 2024-01-15:
I've pushed a fix if you want to start again. Do the octoprint_deploy update or pull the new version.

skewll, 2024-01-16:
Didnt mean to reopen! Sorry