paukstelis/octoprint_deploy #147: Failed to start cam_Ender3S1.service: Unit cam_Ender3S1.service not found.

FireMarshmellow, 2024-01-26:
Hi there, I'm trying to add a camera to an instance. but it keeps getting the Error: 
```
Failed to start cam_Ender3S1.service: Unit cam_Ender3S1.service not found.
Failed to enable unit: Unit file cam_Ender3S1.service does not exist.
```

`*************************
octoprint_deploy 1.0.7
*************************

1) Add instance     3) Add Camera       5) Utilities        7) Update
2) Delete instance  4) Delete Camera    6) Backup Menu      8) Quit
Select operation: 3
Adding camera
1) Ender3S1
2) Ender3pro
3) Quit
Select instance number to add camera to: 1
Selected instance for camera: Ender3S1
Add cameras to haproxy? [y/n]: y

Verify the camera is currently unplugged from USB.....
Is the camera you are trying to detect unplugged from USB? [y/n]: y
Plug your camera in via USB now (detection time-out in 1 min)
/dev/v4l/by-id/usb-Sunplus_IT_Co_HD_720P_webcam_AN20200706001-video-index0
Camera detected with serial number: AN20200706001
No duplicate serial number detected.
Settings can be modified after initial setup in /etc/systemd/system/cam_Ender3S1.service

Camera Resolution [default: 640x480]:

Selected camera resolution: 640x480
Camera Framerate (use 0 for ustreamer hardware) [default: 5]:

Selected camera framerate: 5
cp: cannot stat '/home/OctoprintPI/octoprint_deploy/cam_Ender3S1.service': No such file or directory
Instance must be restarted for settings to take effect. Restart now? [y/n]: y
Failed to start cam_Ender3S1.service: Unit cam_Ender3S1.service not found.
Failed to enable unit: Unit file cam_Ender3S1.service does not exist.`

Any help would be great thank you,

paukstelis, 2024-01-26:
This looks like it is case where a streamer has not been installed. Run octoprint_deploy and provide output from the Utilities->Diagnostic output.

FireMarshmellow, 2024-01-27:
Yep that's probably it, How can i instal it?
```
haproxy: true
streamer: none
octoexec: /home/OctoprintPI/OctoPrint/bin/octoprint
octopip: /home/OctoprintPI/OctoPrint/bin/pip
```

FireMarshmellow, 2024-01-27:
I managed to figure it out, @paukstelis huge thank you for pointing me in the right direction.
For anyone else with the same issue Utilities-> Change streamer.

closed 2024-01-27 by FireMarshmellow