paukstelis/octoprint_deploy #148: Weird issues with obico and camera

Fanatic1074, 2024-01-27:
Im trying to setup two profiles with two cameras.  I have both profiles setup and they work fine.  However I have several issues with the cameras and i know they are not supported but hoping i can get some help.  BTW i am pretty much a newb to this stuff.  Dont know many unix commands.  Please help :-)

First off and this may be the entire issue but i cannot install Ustreamer correctly.  It keeps failing.  If i try to go to utilities and then change streamer and then choose ustreamer i get this....  I think i just need to uninstall and reinstall it or something like that but i dunno how to do that.

Cloning into 'ustreamer'...
remote: Enumerating objects: 179, done.
remote: Counting objects: 100% (179/179), done.
remote: Compressing objects: 100% (130/130), done.
remote: Total 179 (delta 61), reused 108 (delta 37), pack-reused 0
Receiving objects: 100% (179/179), 179.06 KiB | 6.17 MiB/s, done.
Resolving deltas: 100% (61/61), done.
In file included from ustreamer/http/bev.c:23:
ustreamer/http/bev.h:28:10: fatal error: event2/util.h: No such file or directory
 #include <event2/util.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [Makefile:100: build/ustreamer/http/bev.o] Error 1
make[1]: *** [Makefile:38: apps] Error 2
make: *** [Makefile:28: all] Error 2
WARNING! WARNING! WARNING!
Streamer has not been installed correctly.

Also when i look at status on the cameras it says they are failed and shows error codes, here is the output.... both cameras show the same output.

cam_Ender3Max.service - the OctoPi(buntu) ustreamer daemon with the user specified config
   Loaded: loaded (/etc/systemd/system/cam_Ender3Max.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2024-01-27 12:52:15 CST; 1h 26min ago
  Process: 447 ExecStart=/home/pi/ustreamer/ustreamer -d /dev/${DEVICE} -s 0.0.0.0 -m MJPEG -r ${RES} -f ${FRAMERATE} -p ${PORT} --device-timeout 8 --device
 Main PID: 447 (code=exited, status=203/EXEC)

Jan 27 12:52:15 Octopi systemd[1]: Started the OctoPi(buntu) ustreamer daemon with the user specified config.
Jan 27 12:52:15 Octopi systemd[447]: cam_Ender3Max.service: Failed to execute command: No such file or directory
Jan 27 12:52:15 Octopi systemd[447]: cam_Ender3Max.service: Failed at step EXEC spawning /home/pi/ustreamer/ustreamer: No such file or directory
Jan 27 12:52:15 Octopi systemd[1]: cam_Ender3Max.service: Main process exited, code=exited, status=203/EXEC
Jan 27 12:52:15 Octopi systemd[1]: cam_Ender3Max.service: Failed with result 'exit-code'.

Also the cameras are the same make and model camera, when adding the second one it tells me the serial numbers are the same and it may cause issues.  Not sure if that is related.

Obico is only showing one camera and its working but its showing the same cam for both instances.  Something weird is going on.

Thanks!

paukstelis, 2024-01-27:
The WARNING! tells you what has happened. ustreamer has not been installed properly.
Try the following `sudo apt update`
`sudo octoprint_deploy/octoprint_deploy.sh f deb_packages`
Provide the output of that last command.

Fanatic1074, 2024-01-27:
Here is the output.

pi@Octopi:~ $ sudo octoprint_deploy/octoprint_deploy.sh f deb_packages
Reading package lists... Done
Building dependency tree
Reading state information... Done
build-essential is already the newest version (12.6).
g++ is already the newest version (4:8.3.0-1+rpi2).
g++ set to manually installed.
gcc is already the newest version (4:8.3.0-1+rpi2).
gcc set to manually installed.
haproxy is already the newest version (1.8.19-1+rpi1).
libffi-dev is already the newest version (3.2.1-9).
libjpeg62-turbo-dev is already the newest version (1:1.5.2-2+deb10u1).
make is already the newest version (4.2.1-1.2).
make set to manually installed.
pkg-config is already the newest version (0.29-6).
python3-dev is already the newest version (3.7.3-1).
rsync is already the newest version (3.1.3-6).
v4l-utils is already the newest version (1.16.3-3).
cmake is already the newest version (3.16.3-3~bpo10+1).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libjpeg-dev : Conflicts: libjpeg8-dev but 8d1-2 is to be installed
 libjpeg62-turbo-dev : Conflicts: libjpeg8-dev but 8d1-2 is to be installed
E: Unable to correct problems, you have held broken packages.
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package 'brltty' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 247 not upgraded.

paukstelis, 2024-01-27:
I would do a `sudo apt remove libjpeg-dev libjpeg62-turbo-dev` and try the deb_packages command again.

Fanatic1074, 2024-01-27:
Ok i did that and here is the full results. 

pi@Octopi:~ $ sudo apt remove libjpeg-dev libjpeg62-turbo-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package 'libjpeg-dev' is not installed, so not removed
The following packages will be REMOVED:
  libjpeg62-turbo-dev
0 upgraded, 0 newly installed, 1 to remove and 247 not upgraded.
After this operation, 475 kB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 80475 files and directories currently installed.)
Removing libjpeg62-turbo-dev:armhf (1:1.5.2-2+deb10u1) ...
pi@Octopi:~ $ sudo octoprint_deploy/octoprint_deploy.sh f deb_packages
Reading package lists... Done
Building dependency tree
Reading state information... Done
build-essential is already the newest version (12.6).
g++ is already the newest version (4:8.3.0-1+rpi2).
g++ set to manually installed.
gcc is already the newest version (4:8.3.0-1+rpi2).
gcc set to manually installed.
haproxy is already the newest version (1.8.19-1+rpi1).
libffi-dev is already the newest version (3.2.1-9).
make is already the newest version (4.2.1-1.2).
make set to manually installed.
pkg-config is already the newest version (0.29-6).
python3-dev is already the newest version (3.7.3-1).
rsync is already the newest version (3.1.3-6).
v4l-utils is already the newest version (1.16.3-3).
cmake is already the newest version (3.16.3-3~bpo10+1).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libjpeg-dev : Conflicts: libjpeg8-dev but 8d1-2 is to be installed
 libjpeg62-turbo-dev : Conflicts: libjpeg8-dev but 8d1-2 is to be installed
E: Unable to correct problems, you have held broken packages.
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package 'brltty' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 247 not upgraded.

paukstelis, 2024-01-27:
not sure what packages are installed and where the conflict exists.

You can try: `sudo apt --fix-broken`

paukstelis, 2024-01-27:
Ahh....`sudo apt remove libjpeg8-dev` is probably what is needed.

Fanatic1074, 2024-01-27:
Its seems to think its not installed.

pi@Octopi:~ $ sudo apt remove libjpeg8-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package 'libjpeg8-dev' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 247 not upgraded.

paukstelis, 2024-01-27:
I don't know exactly what is going. I've seen similar things like this but there isn't  a clear resolution to me.

try:
`sudo apt install libevent-dev libbsd-dev ffmpeg`

and see if the Streamer install will then work in the utilities menu.

Fanatic1074, 2024-01-27:
Well its a different error LOL.  BTW i really appreciate your assistance here.

Cloning into 'ustreamer'...
remote: Enumerating objects: 179, done.
remote: Counting objects: 100% (179/179), done.
remote: Compressing objects: 100% (130/130), done.
remote: Total 179 (delta 61), reused 108 (delta 37), pack-reused 0
Receiving objects: 100% (179/179), 179.06 KiB | 5.97 MiB/s, done.
Resolving deltas: 100% (61/61), done.
In file included from libs/unjpeg.c:23:
libs/unjpeg.h:33:10: fatal error: jpeglib.h: No such file or directory
 #include <jpeglib.h>
          ^~~~~~~~~~~
compilation terminated.
make[2]: *** [Makefile:100: build/libs/unjpeg.o] Error 1
make[1]: *** [Makefile:38: apps] Error 2
make: *** [Makefile:28: all] Error 2
WARNING! WARNING! WARNING!
Streamer has not been installed correctly.

Fanatic1074, 2024-01-27:
Got streamer to install by installing this....

sudo apt-get install libjpeg8-dev

Now streamer installed right.

Fanatic1074, 2024-01-27:
The cameras still say the same thing.  Do i need to reinstall them now or was that not the issue with the cameras?

Fanatic1074, 2024-01-27:
** Update.  I got rid of all the errors in octoprint_deploy.  

However im getting some strange things still happening.  Printer 1 says no signal and then the instance of printer 2 shows the correct camera first then after about 5 seconds it goes to the other camera feed in a low resolution.  Does this sound like the issue with the duplicate serial number?

Is there a work around for that issue so i can use these two cameras?  They are literally the exact same cam(I was trying to keep consistent)

Thanks!

Fanatic1074, 2024-01-27:
Here is the camera status's.

pi@Octopi:~ $ sudo systemctl status cam_Ender3V2Neo
● cam_Ender3V2Neo.service - the OctoPi(buntu) ustreamer daemon with the user specified config
   Loaded: loaded (/etc/systemd/system/cam_Ender3V2Neo.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2024-01-27 15:52:50 CST; 8min ago
 Main PID: 7672 (main)
    Tasks: 4 (limit: 4915)
   CGroup: /system.slice/cam_Ender3V2Neo.service
           └─7672 /home/pi/ustreamer/ustreamer -d /dev/cam_Ender3V2Neo -s 0.0.0.0 -m MJPEG -r 1920x1080 -f 30 -p 8002 --device-timeout 8 --device-error-delay 8

Jan 27 15:52:50 Octopi ustreamer[7672]: -- INFO  [10826.562    stream] -- Requested 5 device buffers, got 5
Jan 27 15:52:51 Octopi ustreamer[7672]: -- INFO  [10826.752    stream] -- Capturing started
Jan 27 15:52:51 Octopi ustreamer[7672]: -- INFO  [10826.752    stream] -- Switching to HW encoder: the input is (M)JPEG ...
Jan 27 15:52:51 Octopi ustreamer[7672]: -- INFO  [10826.752    stream] -- Using JPEG quality: encoder default
Jan 27 15:52:51 Octopi ustreamer[7672]: -- INFO  [10826.752    stream] -- Creating pool JPEG with 1 workers ...
Jan 27 15:52:51 Octopi ustreamer[7672]: -- INFO  [10826.752    stream] -- Capturing ...
Jan 27 15:53:36 Octopi ustreamer[7672]: -- INFO  [10872.519      http] -- HTTP: NEW client (now=1): [::ffff:192.168.1.224]:56672, id=817d8aaf2f5561c3
Jan 27 15:54:10 Octopi ustreamer[7672]: -- INFO  [10906.265      http] -- HTTP: DEL client (now=0): [::ffff:192.168.1.224]:56672, id=817d8aaf2f5561c3, Connection reset by peer (reading,error)
Jan 27 15:54:12 Octopi ustreamer[7672]: -- INFO  [10907.874      http] -- HTTP: NEW client (now=1): [::ffff:192.168.1.224]:56854, id=8e6f2465b479b937
Jan 27 15:54:21 Octopi ustreamer[7672]: -- INFO  [10917.041      http] -- HTTP: DEL client (now=0): [::ffff:192.168.1.224]:56854, id=8e6f2465b479b937, Resource temporarily unavailable (reading,eof)
pi@Octopi:~ $ sudo systemctl restart cam_Ender3Max
pi@Octopi:~ $ sudo systemctl status cam_Ender3Max
● cam_Ender3Max.service - the OctoPi(buntu) ustreamer daemon with the user specified config
   Loaded: loaded (/etc/systemd/system/cam_Ender3Max.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2024-01-27 16:01:28 CST; 4s ago
 Main PID: 14291 (main)
    Tasks: 3 (limit: 4915)
   CGroup: /system.slice/cam_Ender3Max.service
           └─14291 /home/pi/ustreamer/ustreamer -d /dev/cam_Ender3Max -s 0.0.0.0 -m MJPEG -r 1920x1080 -f 30 -p 8001 --device-timeout 8 --device-error-delay 8

Jan 27 16:01:28 Octopi ustreamer[14291]: -- INFO  [11344.111    stream] -- Using V4L2 device: /dev/cam_Ender3Max
Jan 27 16:01:28 Octopi ustreamer[14291]: -- INFO  [11344.111    stream] -- Using desired FPS: 30
Jan 27 16:01:28 Octopi ustreamer[14291]: -- INFO  [11344.111      http] -- Starting HTTP eventloop ...
Jan 27 16:01:28 Octopi ustreamer[14291]: ================================================================================
Jan 27 16:01:28 Octopi ustreamer[14291]: -- INFO  [11344.112    stream] -- Device fd=8 opened
Jan 27 16:01:28 Octopi ustreamer[14291]: -- INFO  [11344.112    stream] -- Using capture type: single-planar
Jan 27 16:01:28 Octopi ustreamer[14291]: -- INFO  [11344.112    stream] -- Using input channel: 0
Jan 27 16:01:28 Octopi ustreamer[14291]: -- ERROR [11344.112    stream] -- Can't set input channel
Jan 27 16:01:28 Octopi ustreamer[14291]: -- INFO  [11344.112    stream] -- Device fd=8 closed
Jan 27 16:01:28 Octopi ustreamer[14291]: -- INFO  [11344.112    stream] -- Sleeping 8 seconds before new stream init ...
pi@Octopi:~ $

paukstelis, 2024-01-28:
best to run octoprint_deploy and provide output from Utilities -> Diagnostic output.

Fanatic1074, 2024-01-28:
Yea i did try to delete and re-add the cameras but the issue just moved to the other cam.  Here is the output from diags.

octoprint_deploy diagnostic information. Please provide ALL output for support help
**************************************
/etc/octoprint_deploy
**************************************
octoexec: /home/pi/oprint/bin/octoprint
octopip: /home/pi/oprint/bin/pip
haproxy: true
octopi: true
streamer: ustreamer
**************************************
/etc/octoprint_instances
**************************************
instance:octoprint port:5000 udev:false
instance:Ender3Max port:5001 udev:true
instance:Ender3V2Neo port:5002 udev:true
**************************************
/etc/octoprint_cameras
**************************************
camera:cam_Ender3Max port:8001 udev:true
camera:cam_Ender3V2Neo port:8002 udev:true
**************************************
/etc/udev/rules.d/99-octoprint.rules
**************************************
KERNELS=="1-1.1:1.0",SUBSYSTEM=="tty",SYMLINK+="octo_Ender3Max"
KERNELS=="1-1.2:1.0",SUBSYSTEM=="tty",SYMLINK+="octo_Ender3V2Neo"
SUBSYSTEM=="video4linux", ATTRS{serial}=="SN0001", ATTR{index}=="0", SYMLINK+="cam_Ender3Max"
SUBSYSTEM=="video4linux", ATTRS{serial}=="SN0001", ATTR{index}=="0", SYMLINK+="cam_Ender3V2Neo"
lrwxrwxrwx 1 root root 7 Jan 27 16:46 /dev/octo_Ender3Max -> ttyUSB0
lrwxrwxrwx 1 root root 6 Jan 27 16:46 /dev/cam_Ender3Max -> video0
lrwxrwxrwx 1 root root 6 Jan 27 16:46 /dev/cam_Ender3V2Neo -> video0
**************************************
● octoprint.service - The snappy web interface for your 3D printer
   Loaded: loaded (/etc/systemd/system/octoprint.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2024-01-27 16:28:21 CST; 3h 26min ago
 Main PID: 448 (octoprint)
    Tasks: 22 (limit: 4915)
   CGroup: /system.slice/octoprint.service
           └─448 /home/pi/oprint/bin/python3 /home/pi/oprint/bin/octoprint serve --host=127.0.0.1 --port=5000

Jan 27 19:05:21 Octopi octoprint[448]: 2024-01-27 19:05:21,947 - backoff - INFO - Backing off wait_for_auth_token(...) for 241.8s (None)
Jan 27 19:09:23 Octopi octoprint[448]: 2024-01-27 19:09:23,980 - backoff - INFO - Backing off wait_for_auth_token(...) for 763.0s (None)
Jan 27 19:13:43 Octopi octoprint[448]: 2024-01-27 19:13:43,795 - octoprint.server.heartbeat - INFO - Server heartbeat <3
Jan 27 19:22:07 Octopi octoprint[448]: 2024-01-27 19:22:07,154 - backoff - INFO - Backing off wait_for_auth_token(...) for 71.5s (None)
Jan 27 19:23:18 Octopi octoprint[448]: 2024-01-27 19:23:18,873 - backoff - INFO - Backing off wait_for_auth_token(...) for 599.6s (None)
Jan 27 19:28:43 Octopi octoprint[448]: 2024-01-27 19:28:43,796 - octoprint.server.heartbeat - INFO - Server heartbeat <3
Jan 27 19:33:18 Octopi octoprint[448]: 2024-01-27 19:33:18,706 - backoff - INFO - Backing off wait_for_auth_token(...) for 538.2s (None)
Jan 27 19:42:17 Octopi octoprint[448]: 2024-01-27 19:42:17,138 - backoff - INFO - Backing off wait_for_auth_token(...) for 635.8s (None)
Jan 27 19:43:43 Octopi octoprint[448]: 2024-01-27 19:43:43,797 - octoprint.server.heartbeat - INFO - Server heartbeat <3
Jan 27 19:52:53 Octopi octoprint[448]: 2024-01-27 19:52:53,113 - backoff - INFO - Backing off wait_for_auth_token(...) for 231.7s (None)
None
None
{'bitrate': '10000k',
 'cleanTmpAfterDays': 7,
 'defaultWebcam': 'classic',
 'ffmpeg': None,
 'ffmpegCommandline': '{ffmpeg} -framerate {fps} -i "{input}" -vcodec '
                      '{videocodec} -threads {threads} -b:v {bitrate} -f '
                      '{containerformat} -y {filters} "{output}"',
 'ffmpegThreads': 1,
 'ffmpegThumbnailCommandline': '{ffmpeg} -sseof -1 -i "{input}" -update 1 -q:v '
                               '0.7 "{output}"',
 'ffmpegVideoCodec': 'libx264',
 'snapshotWebcam': 'classic',
 'timelapse': {'fps': 25,
               'options': {'capturePostRoll': None,
                           'interval': None,
                           'retractionZHop': None},
               'postRoll': 0,
[... 130 lines cut ...]
Jan 27 19:55:04 Octopi ustreamer[14150]: -- ERROR [12393.880    stream] -- Can't set input channel
Jan 27 19:55:04 Octopi ustreamer[14150]: -- INFO  [12393.880    stream] -- Device fd=9 closed
Jan 27 19:55:04 Octopi ustreamer[14150]: -- INFO  [12393.880    stream] -- Sleeping 8 seconds before new stream init ...
**************************************
● haproxy.service - HAProxy Load Balancer
   Loaded: loaded (/lib/systemd/system/haproxy.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2024-01-27 16:46:15 CST; 3h 8min ago
     Docs: man:haproxy(1)
           file:/usr/share/doc/haproxy/configuration.txt.gz
 Main PID: 14143 (haproxy)
    Tasks: 2 (limit: 4915)
   CGroup: /system.slice/haproxy.service
           ├─14143 /usr/sbin/haproxy -Ws -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid
           └─14147 /usr/sbin/haproxy -Ws -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid

Jan 27 19:54:51 Octopi haproxy[14147]: ::ffff:192.168.1.175:36022 [27/Jan/2024:19:54:51.195] public octoprint/octoprint1 0/0/0/44/44 200 962 - - ---- 7/7/1/1/0 0/0 "GET /plugin/discovery/discovery.xml HTTP/1.1"
Jan 27 19:54:51 Octopi haproxy[14147]: ::ffff:192.168.1.175:36028 [27/Jan/2024:19:54:51.207] public octoprint/octoprint1 0/0/1/47/48 200 962 - - ---- 6/6/0/0/0 0/0 "GET /plugin/discovery/discovery.xml HTTP/1.1"
[... cut ...]
Jan 27 19:55:02 Octopi haproxy[14147]: ::ffff:192.168.1.241:43253 [27/Jan/2024:19:55:02.719] public octoprint/octoprint1 0/0/0/15/16 200 962 - - ---- 6/6/0/0/0 0/0 "GET /plugin/discovery/discovery.xml HTTP/1.1"
Jan 27 19:55:05 Octopi haproxy[14147]: ::ffff:192.168.1.241:43035 [27/Jan/2024:19:55:05.611] public octoprint/octoprint1 0/0/0/14/14 200 962 - - ---- 5/5/0/0/0 0/0 "GET /plugin/discovery/discovery.xml HTTP/1.1"
Jan 27 19:55:07 Octopi haproxy[14147]: ::ffff:192.168.1.2:1794 [27/Jan/2024:19:55:07.708] public octoprint/octoprint1 0/0/0/14/14 200 962 - - ---- 5/5/0/0/0 0/0 "GET /plugin/discovery/discovery.xml HTTP/1.1"

paukstelis, 2024-01-28:
Yes, your camera issue is from duplicate serial numbers.
The way around this is to remove both cameras.
Re-run octoprint_deploy with the noserial option (sudo octoprint_deploy/octoprint_deploy.sh noserial) and then add the cameras back.

Fanatic1074, 2024-01-28:
Excellent.  I will try that after my print finishes.  

Thanks for your help paul.  Do you have a venmo or something i can donate to?  I dont have much but i would like to show my appreciation.  

Thanks!

paukstelis, 2024-01-28:
You are quite welcome. I need to investigate what the issue is with OctoPi and the packages it is using. There only ever seems to be an issue with OctoPi.

There is a buymeacoffee link in the README. Venmo is @Paul-Paukstelis

Fanatic1074, 2024-01-28:
Ok, bought you a few coffess lol. That seems to have fixed the octoprint issues.  Any idea why obico keeps flashing and mixing up the cameras?  I know that might be out of your wheelhouse.

paukstelis, 2024-01-28:
Have you done a reboot since you fixed the issue in octoprint_deploy? You may need to just restart the instances for obico to be happy.

thanks for the coffee!

Fanatic1074, 2024-01-28:
I did reboot.  I just started another print.  I have several things i need to get done before the end of the weekend but, i think i will start over with obico when this one gets done and see if that helps.  obico is pretty much useless like this.  It just keeps going from online to offline literally once a second.

Fanatic1074, 2024-06-17:
Ok im finally back, life got in the way and i just stopped printing for a while.  

So in order to try to fix the issue i have completely started over.  Put a new fresh image of Raspberry Pi on my machine and redone all the octoprint stuff.  now i am getting a strange "activating" next to each camera status.  Any idea why the camera would be stuck in activating status?

Fanatic1074, 2024-06-17:
This is where i seem to get errors starting if i use the diag output.

Jun 17 07:55:48 RaspberryPi octoprint[378]: 2024-06-17 07:55:48,372 - backoff - ERROR - Giving up capture_jpeg(...) after 3 tries (requests.exceptions.ConnectionError: HTTPConnectionPool(host='localhost', port=8001): Max retries exceeded with url: /?action=snapshot (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x9c8b85f8>: Failed to establish a new connection: [Errno 111] Connection refused')))
Jun 17 07:55:48 RaspberryPi octoprint[378]: 2024-06-17 07:55:48,532 - octoprint.plugins.obico - WARNING - Failed to capture jpeg - HTTPConnectionPool(host='localhost', port=8001): Max retries exceeded with url: /?action=snapshot (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x9c8b85f8>: Failed to establish a new connection: [Errno 111] Connection refused'))
Jun 17 07:55:48 RaspberryPi octoprint[378]: 2024-06-17 07:55:48,538 - backoff - INFO - Backing off capture_jpeg(...) for 0.6s (requests.exceptions.ConnectionError: HTTPConnectionPool(host='localhost', port=8001): Max retries exceeded with url: /?action=snapshot (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xa72b4688>: Failed to establish a new connection: [Errno 111] Connection refused')))
Jun 17 07:55:49 RaspberryPi octoprint[378]: 2024-06-17 07:55:49,118 - backoff - INFO - Backing off capture_jpeg(...) for 1.3s (requests.exceptions.ConnectionError: HTTPConnectionPool(host='localhost', port=8001): Max retries exceeded with url: /?action=snapshot (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xa6fe5df0>: Failed to establish a new connection: [Errno 111] Connection refused')))
Jun 17 07:55:50 RaspberryPi octoprint[378]: 2024-06-17 07:55:50,473 - backoff - ERROR - Giving up capture_jpeg(...) after 3 tries (requests.exceptions.ConnectionError: HTTPConnectionPool(host='localhost', port=8001): Max retries exceeded with url: /?action=snapshot (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x9c8de568>: Failed to establish a new connection: [Errno 111] Connection refused')))
Jun 17 07:55:50 RaspberryPi octoprint[378]: 2024-06-17 07:55:50,474 - octoprint.plugins.obico - WARNING - Failed to capture jpeg - HTTPConnectionPool(host='localhost', port=8001): Max retries exceeded with url: /?action=snapshot (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x9c8de568>: Failed to establish a new connection: [Errno 111] Connection refused'))
Jun 17 08:09:59 RaspberryPi octoprint[378]: 2024-06-17 08:09:59,213 - octoprint.server.heartbeat - INFO - Server heartbeat <3
Jun 17 08:24:59 RaspberryPi octoprint[378]: 2024-06-17 08:24:59,214 - octoprint.server.heartbeat - INFO - Server heartbeat <3
Jun 17 08:39:59 RaspberryPi octoprint[378]: 2024-06-17 08:39:59,216 - octoprint.server.heartbeat - INFO - Server heartbeat <3
Jun 17 08:54:59 RaspberryPi octoprint[378]: 2024-06-17 08:54:59,217 - octoprint.server.heartbeat - INFO - Server heartbeat <3

paukstelis, 2024-06-17:
It looks like you need to modify the snapshot address in the camera settings. You can try 127.0.0.1:8001/?action=snapshot

Fanatic1074, 2024-06-17:
Well i came to a very similar conclusion,  I noticed that localhost was returning an invalid ip v6 address, so i just disabled ip v6 and now im getting a different error about ffmpeg.  However i am pretty sure its installed ok.  see below.

 Jun 17 10:38:00 RaspberryPi octoprint[420]: STDERR:
Jun 17 10:38:00 RaspberryPi octoprint[420]: b'http://localhost/cam_Ender3V2Neo/?action=stream: Server returned 5XX Server Error reply\n'
Jun 17 10:38:00 RaspberryPi octoprint[420]:
Jun 17 10:38:00 RaspberryPi octoprint[420]: 2024-06-17 10:38:00,370 - octoprint.plugins.obico - ERROR - Exception
Jun 17 10:38:00 RaspberryPi octoprint[420]: Traceback (most recent call last):
Jun 17 10:38:00 RaspberryPi octoprint[420]:   File "/home/Fanatic/oprint/lib/python3.9/site-packages/octoprint_obico/webcam_stream.py", line 379, in h264_transcode
Jun 17 10:38:00 RaspberryPi octoprint[420]:     self.start_ffmpeg(rtp_port, '-re -i {stream_url} -filter:v fps={fps} -b:v {bitrate} -pix_fmt yuv420p -s {img_w}x{img_h} {encoder}'.format(stream_url=stream_url, fps=fps, bitrate=bitrate, img_w=img_w, img_h=img_h, encoder=webcam['streaming_params'].get('h264_encoder')))
Jun 17 10:38:00 RaspberryPi octoprint[420]:   File "/home/Fanatic/oprint/lib/python3.9/site-packages/octoprint_obico/webcam_stream.py", line 401, in start_ffmpeg
Jun 17 10:38:00 RaspberryPi octoprint[420]:     raise Exception('ffmpeg failed! Exit code: {}'.format(returncode))
Jun 17 10:38:00 RaspberryPi octoprint[420]: Exception: ffmpeg failed! Exit code: 1
2024-06-17 10:39:34,159 - octolapse.__init__ - INFO - Release mode detected.

paukstelis, 2024-06-17:
Did you put in the path to ffmpeg?
You can get that from terminal with `which ffmpeg`

Fanatic1074, 2024-06-17:
It was along night last night but i believe i verified that.  This is what i get

/usr/bin/ffmpeg

But i forget where to put it...

Fanatic1074, 2024-06-17:
Ok, found it, its good.

[image]

Fanatic1074, 2024-06-18:
Ok i am completely lost now.  It looks like im not even getting errors but camera status's are still activating and i got no stream.

Help Please.

Here is the Diag output.

octoprint_deploy diagnostic information. Please provide ALL output for support help
**************************************
/etc/octoprint_deploy
**************************************
octoexec: /home/Fanatic/oprint/bin/octoprint
octopip: /home/Fanatic/oprint/bin/pip
haproxy: true
octopi: true
streamer: camera-streamer
**************************************
/etc/octoprint_instances
**************************************
instance:Ender-3-Max port:5000 udev:true
instance:Ender-3-V2-Neo port:5001 udev:true
**************************************
/etc/octoprint_cameras
**************************************
camera:cam_Ender-3-Max port:8001 udev:true
camera:cam_Ender-3-V2-Neo port:8002 udev:true
**************************************
/etc/udev/rules.d/99-octoprint.rules
**************************************
KERNELS=="1-1.1.4:1.0",SUBSYSTEM=="tty",SYMLINK+="octo_Ender-3-Max"
SUBSYSTEM=="video4linux",KERNELS=="1-1.1.3:1.0", SUBSYSTEMS=="usb", ATTR{index}=="0", DRIVERS=="uvcvideo", SYMLINK+="cam_Ender-3-Max"
KERNELS=="1-1.1.2:1.0",SUBSYSTEM=="tty",SYMLINK+="octo_Ender-3-V2-Neo"
SUBSYSTEM=="video4linux",KERNELS=="1-1.1.1.4:1.0", SUBSYSTEMS=="usb", ATTR{index}=="0", DRIVERS=="uvcvideo", SYMLINK+="cam_Ender-3-V2-Neo"
lrwxrwxrwx 1 root root 7 Jun 18 14:28 /dev/octo_Ender-3-Max -> ttyUSB1
lrwxrwxrwx 1 root root 7 Jun 18 14:28 /dev/octo_Ender-3-V2-Neo -> ttyUSB0
lrwxrwxrwx 1 root root 6 Jun 18 14:28 /dev/cam_Ender-3-Max -> video0
lrwxrwxrwx 1 root root 6 Jun 18 14:28 /dev/cam_Ender-3-V2-Neo -> video2
**************************************
● Ender-3-Max.service - The snappy web interface for your 3D printer
     Loaded: loaded (/etc/systemd/system/Ender-3-Max.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2024-06-18 14:28:13 CDT; 1min 55s ago
   Main PID: 398 (octoprint)
      Tasks: 22 (limit: 4915)
        CPU: 17.065s
     CGroup: /system.slice/Ender-3-Max.service
             └─398 /home/Fanatic/oprint/bin/python /home/Fanatic/oprint/bin/octoprint serve --config=/home/Fanatic/.Ender-3-Max/config.yaml --basedir=/home/Fanatic/.Ender-3-Max --port=5000

Jun 18 14:28:27 RaspberryPi octoprint[398]: 2024-06-18 14:28:27,661 - octoprint.plugins.discovery - INFO - Registered 'OctoPrint instance "Ender-3-Max"._octoprint._tcp.local.' for _octoprint._tcp.local.
Jun 18 14:28:27 RaspberryPi octoprint[398]: 2024-06-18 14:28:27,690 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance "Ender-3-Max" for SSDP
Jun 18 14:28:27 RaspberryPi octoprint[398]: 2024-06-18 14:28:27,708 - octoprint.server - INFO - Listening on http://0.0.0.0:5000 and http://[::]:5000
Jun 18 14:28:27 RaspberryPi octoprint[398]: 2024-06-18 14:28:27,837 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
Jun 18 14:28:29 RaspberryPi octoprint[398]: 2024-06-18 14:28:29,298 - octoprint.util.pip - INFO - Using "/home/Fanatic/oprint/bin/python -m pip" as command to invoke pip
Jun 18 14:28:29 RaspberryPi octoprint[398]: 2024-06-18 14:28:29,310 - octoprint.util.pip - INFO - pip installs to /home/Fanatic/oprint/lib/python3.9/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
Jun 18 14:28:29 RaspberryPi octoprint[398]: 2024-06-18 14:28:29,323 - octoprint.util.pip - INFO - ==> pip ok -> yes
Jun 18 14:28:29 RaspberryPi octoprint[398]: 2024-06-18 14:28:29,401 - octoprint.plugins.softwareupdate - INFO - Minimum free storage across all update relevant locations is 52.7GB. That is considered sufficient for updating.
Jun 18 14:28:29 RaspberryPi octoprint[398]: 2024-06-18 14:28:29,402 - octoprint.plugins.softwareupdate - INFO - Fetching check overlays from https://plugins.octoprint.org/update_check_overlay.json
Jun 18 14:28:29 RaspberryPi octoprint[398]: 2024-06-18 14:28:29,820 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
None
None
{'bitrate': '10000k',
 'cleanTmpAfterDays': 7,
 'defaultWebcam': 'classic',
 'ffmpeg': None,
 'ffmpegCommandline': '{ffmpeg} -framerate {fps} -i "{input}" -vcodec '
                      '{videocodec} -threads {threads} -b:v {bitrate} -f '
                      '{containerformat} -y {filters} "{output}"',
 'ffmpegThreads': 1,
 'ffmpegThumbnailCommandline': '{ffmpeg} -sseof -1 -i "{input}" -update 1 -q:v '
                               '0.7 "{output}"',
 'ffmpegVideoCodec': 'libx264',
 'snapshotWebcam': 'classic',
[... 65 lines cut ...]
        CPU: 4ms
**************************************
● haproxy.service - HAProxy Load Balancer
     Loaded: loaded (/lib/systemd/system/haproxy.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2024-06-18 14:28:24 CDT; 2min 1s ago
       Docs: man:haproxy(1)
             file:/usr/share/doc/haproxy/configuration.txt.gz
    Process: 1105 ExecStartPre=/usr/sbin/haproxy -f $CONFIG -c -q $EXTRAOPTS (code=exited, status=0/SUCCESS)
   Main PID: 1115 (haproxy)
      Tasks: 5 (limit: 4915)
        CPU: 244ms
     CGroup: /system.slice/haproxy.service
             ├─1115 /usr/sbin/haproxy -Ws -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid -S /run/haproxy-master.sock
             └─1125 /usr/sbin/haproxy -Ws -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid -S /run/haproxy-master.sock

[... cut ...]
Jun 18 14:30:23 RaspberryPi haproxy[1125]: ::ffff:192.168.1.224:14315 [18/Jun/2024:14:30:23.390] public octoprint/octoprint1 0/0/0/19/19 403 300 - - ---- 6/6/0/0/0 0/0 "GET /api/job HTTP/1.1"
Jun 18 14:30:25 RaspberryPi haproxy[1125]: ::ffff:192.168.1.224:14314 [18/Jun/2024:14:30:25.385] public octoprint/octoprint1 0/0/0/19/19 403 300 - - ---- 6/6/1/1/0 0/0 "GET /api/printer HTTP/1.1"
Jun 18 14:30:25 RaspberryPi haproxy[1125]: ::ffff:192.168.1.224:14315 [18/Jun/2024:14:30:25.385] public octoprint/octoprint1 0/0/0/20/20 403 300 - - ---- 6/6/0/0/0 0/0 "GET /api/job HTTP/1.1"

Fanatic1074, 2024-06-21:
Can anyone help with this im out of ideas.  Thanks!

paukstelis, 2024-06-21:
Remove cameras from the menu and change the streamer (in utility menu) to ustreamer. Re-add the cameras and see if that works for you.

Fanatic1074, 2024-06-21:
Well i got it but i think it was another FFMPEG issue.  When i removed the cameras and tried to change streamers i got another FFMPEG error.  So i completely removed FFMPEG and reinstalled it, added the cameras back and it worked!  I think maybe the image of the OctoPi i got has a corrupt FFMPEG install.  OR... maybe it was re adding them i dunno for sure.  But its working.  Thank you!