paukstelis/octoprint_deploy #150: High CPU usage with camera

JonSnow88, 2024-01-29:
I'm running octoprint on proxmox installed in a debain VM, I needed to give the VM 2 cores and its still at about 50% usage doing nothing. 

I enabled the default ustreamer option with 0 during setup, how can I get the CPU usage down?

If the reason is because of software decoding in a VM, then what is the command to disable the camera feed?

On my previous setup, whenever I powered on/off my wifi plugs they issue a command to turn on/off the camera stream.

paukstelis, 2024-01-29:
I can't tell you why it is high CPU usage, though if the camera doesn't have MJPEG compression it will use a lot more CPU.
To disable a camera: `sudo systemctl disable cam_instancename` and `sudo systemctl stop cam_instancename`

JonSnow88, 2024-01-29:
It's a Wyze Cam v2 with webcam firmware, have used this for a few years on a Pi and have't seen high CPU usage on it.

I've tried passing the Intel integrated GPU to the VM but it has had no effect on CPU, is there a way to enable hardware decoding for ustreamer?

Wyze Cam is supported:
https://community.octoprint.org/t/usb-webcams-known-to-work-with-mjpg-streamer/21149

paukstelis, 2024-01-29:
ustreamer will use hardware decode by default.

I don't use cameras at all, so I'm afraid I won't be able to give you any insight on this issue.

JonSnow88, 2024-01-29:
I may attempt to reinstall debian, with iGPU pass though before install.

That may have been my mistake