paukstelis/octoprint_deploy #153: No camera detected

CH-MrMaus, 2024-02-03:
I have this strange problem, where there is no camera detected.

The System is a Ubuntu Server 22.04, running in a proxmox privileged LCX. The instance Does work properly, and I can print, but when trying to add a camera it will not be detected. 

One cam has this output of usb-devices

> T:  Bus=03 Lev=01 Prnt=01 Port=01 Cnt=01 Dev#= 13 Spd=480 MxCh= 0
> D:  Ver= 2.00 Cls=ef(misc ) Sub=02 Prot=01 MxPS=64 #Cfgs=  1
> P:  Vendor=1b3f ProdID=2247 Rev=01.00
> S:  Manufacturer=GENERAL
> S:  Product=GENERAL WEBCAM
> C:  #Ifs= 4 Cfg#= 1 Atr=80 MxPwr=100mA
> I:  If#= 0 Alt= 0 #EPs= 1 Cls=0e(video) Sub=01 Prot=00 Driver=uvcvideo
> E:  Ad=83(I) Atr=03(Int.) MxPS=  64 Ivl=32ms
> I:  If#= 1 Alt= 0 #EPs= 0 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
> I:  If#= 2 Alt= 0 #EPs= 0 Cls=01(audio) Sub=01 Prot=00 Driver=snd-usb-audio
> I:  If#= 3 Alt= 0 #EPs= 0 Cls=01(audio) Sub=02 Prot=00 Driver=snd-usb-audio

the second this one:

> T:  Bus=03 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#= 14 Spd=480 MxCh= 0
> D:  Ver= 2.00 Cls=ef(misc ) Sub=02 Prot=01 MxPS=64 #Cfgs=  1
> P:  Vendor=328f ProdID=2015 Rev=06.11
> S:  Manufacturer=eMeet-200611
> S:  Product=eMeet Nova
> C:  #Ifs= 4 Cfg#= 1 Atr=80 MxPwr=500mA
> I:  If#= 0 Alt= 0 #EPs= 1 Cls=0e(video) Sub=01 Prot=00 Driver=uvcvideo
> E:  Ad=87(I) Atr=03(Int.) MxPS=  16 Ivl=16ms
> I:  If#= 1 Alt= 0 #EPs= 0 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
> I:  If#= 2 Alt= 0 #EPs= 0 Cls=01(audio) Sub=01 Prot=00 Driver=snd-usb-audio
> I:  If#= 3 Alt= 0 #EPs= 0 Cls=01(audio) Sub=02 Prot=00 Driver=snd-usb-audio

Thanks for haveing a look

paukstelis, 2024-02-03:
When I briefly looked at LCX I came across issues with integrating with peripherals (though I think that was mainly on making udev rules).

Two things to do:
1.Make sure you are plugging the camera in when it tells you to (verify it is unplugged, y, plug in camera)

2. You can see what dmesg sees for your camera: `sudo dmesg -C` plug in camera, `sudo dmesg`

CH-MrMaus, 2024-02-03:
As long as u use a privileged container it should actually work fine, but there are risks in doing that... 

Sure, the first is for the generic the second for the nova..

> 
> [88579.865005] usb 1-2: new high-speed USB device number 10 using xhci_hcd
> [88580.014213] usb 1-2: config 1 interface 0 altsetting 0 endpoint 0x83 has an i                                          nvalid bInterval 32, changing to 9
> [88580.014520] usb 1-2: New USB device found, idVendor=1b3f, idProduct=2247, bcd                                          Device= 1.00
> [88580.014528] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
> [88580.014533] usb 1-2: Product: GENERAL WEBCAM
> [88580.014536] usb 1-2: Manufacturer: GENERAL
> [88580.015775] usb 1-2: Found UVC 1.00 device GENERAL WEBCAM (1b3f:2247)
> [88580.018715] usb 1-2: UVC non compliance - GET_DEF(PROBE) not supported. Enabl                                          ing workaround.
> [88580.028609] usb 1-2: Warning! Unlikely big volume range (=5120), cval->res is                                           probably wrong.
> [88580.028626] usb 1-2: [5] FU [Mic Capture Volume] ch = 1, val = 7680/12800/1

and

> [88695.601491] usb 1-2: new high-speed USB device number 11 using xhci_hcd
> [88696.051485] usb 1-2: New USB device found, idVendor=328f, idProduct=2015, bcdDevice= 6.11
> [88696.051504] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
> [88696.051508] usb 1-2: Product: eMeet Nova
> [88696.051530] usb 1-2: Manufacturer: eMeet-200611
> [88696.274223] usb 1-2: Found UVC 1.00 device eMeet Nova (328f:2015)
> [88696.371797] usb 1-2: Warning! Unlikely big volume range (=2816), cval->res is probably wrong.
> [88696.371829] usb 1-2: [5] FU [Mic Capture Volume] ch = 1, val = 0/2816/1
>

Thanks

paukstelis, 2024-02-05:
The cameras are not providing any of the 3 pieces of information that octoprint_deploy uses for auto detection, it would seem. It will identify the serial number, if available, the USB port, or the by-id address. Not sure why that is the case. Might be related to LCX, or might just be the cameras themselves.

CH-MrMaus, 2024-02-06:
Perfect!

So cameras generally might not work wit Proxmox LCX.

closed 2024-02-06 by CH-MrMaus