paukstelis/octoprint_deploy #156: µstreamer fails to install

beelsebob, 2024-02-28:
With a completely fresh install of rpios µstreamer fails to install correctly:

```Which video streamer you would like to install?: 1
Cloning into 'ustreamer'...
remote: Enumerating objects: 188, done.
remote: Counting objects: 100% (188/188), done.
remote: Compressing objects: 100% (141/141), done.
remote: Total 188 (delta 66), reused 103 (delta 35), pack-reused 0
Receiving objects: 100% (188/188), 191.05 KiB | 2.39 MiB/s, done.
Resolving deltas: 100% (66/66), done.
WARNING! WARNING! WARNING!
Streamer has not been installed correctly.```

beelsebob, 2024-02-28:
Looks like it was caused by https://github.com/pikvm/ustreamer/commit/3715c89ec8a5465d88ee672be28ad65c59c4599d ustreamer and ustreamer-dump are no longer created as links to ustreamer.bin and ustreamer-dump.bin.

paukstelis, 2024-02-29:
This is now fixed in 1.0.9.

closed 2024-02-29 by paukstelis