paukstelis/octoprint_deploy #162: [Feature] Camera: configurable pixel format (for ustreamer)

dancesWithMachines, 2024-04-01:
Hi,
I recently set up octoprint via octoprint_deploy and everything went great except camera, it added successfully but I was getting "no signal" on display. My problem was that my camera (PS3 eye) does not support MJPEG pixel format for ustreamer, and I got to manually change it for YUYV in camera service file.

From what I established, pixel format is not configurable via octoprint_deploy add camera dialog. It would be nice, to let user choose pixel format in "add camera" dialog leaving MJPEG as default. My motivation is that users might be able to fix similar issues with trial and error method or by looking up their camera's output format, rather than learn how it works under the hood.

paukstelis, 2024-04-17:
YUYV cameras are terrible and outdated. It is an extra step that 98% of users won't even know what setting their cameras need. Defaulting to MJPG makes more sense IMO.

dancesWithMachines, 2024-04-18:
They might be outdated but are still widely used (said ps3 eye for example). I agree that MJPG should be default, but with option to change it. I don't expect users to know what setting they need, I'm just saying if this option is made configurable, users are more likely to look it up (in case if it's not working) or use trial and error method to make it work.