paukstelis/octoprint_deploy #164: Webcam doesn't work

Blisk, 2024-04-16:
After installing octoprint_deploy web camera stops working.
I have made a fresh install of raspberry pi and when I did that I get camera picture.
After installing deploy, camera doesn't work anymore.
I user raspberry pi 4 and it is fully updated.

paukstelis, 2024-04-17:
You have to provide more information. Please re-run octoprint_deploy and use Utilties->Diagnostic Output.

ingo916, 2024-05-04:
*************************
octoprint_deploy 1.0.11
*************************

1) Add instance 3) Add USB Camera 5) Delete Camera 7) Backup Menu 9) Quit
2) Delete instance 4) Add PiCam 6) Utilities 8) Update
Select operation: 6

1) Instance Status 3) Sync Users 5) Change Streamer 7) Udev Menu 9) Quit
2) USB Port Testing 4) Share Uploads 6) Set Global Config 8) Diagnostic Output
Select an option: 8
octoprint_deploy diagnostic information. Please provide ALL output for support help
**************************************
/etc/octoprint_deploy
**************************************
haproxy: true
octoexec: /home/pi/OctoPrint/bin/octoprint
octopip: /home/pi/OctoPrint/bin/pip
streamer: camera-streamer
**************************************
/etc/octoprint_instances
**************************************
instance:Ender3v2 port:5000 udev:true
**************************************
/etc/octoprint_cameras
**************************************
camera:cam_Ender3v2 port:8001 udev:true
**************************************
/etc/udev/rules.d/99-octoprint.rules
**************************************
KERNELS=="3-1:1.0",SUBSYSTEM=="tty",SYMLINK+="octo_Ender3v2"
SUBSYSTEM=="video4linux", ATTRS{serial}=="43E9B72F", ATTR{index}=="0", SYMLINK+="cam_Ender3v2"
lrwxrwxrwx 1 root root 7 May 3 13:36 /dev/octo_Ender3v2 -> ttyUSB0
lrwxrwxrwx 1 root root 6 May 3 13:36 /dev/cam_Ender3v2 -> video0
**************************************
● Ender3v2.service - The snappy web interface for your 3D printer
 Loaded: loaded (/etc/systemd/system/Ender3v2.service; enabled; preset: enabled)
 Active: active (running) since Fri 2024-05-03 13:36:42 PDT; 7h ago
 Main PID: 763 (octoprint)
 Tasks: 43 (limit: 9247)
 CPU: 1min 16.419s
 CGroup: /system.slice/Ender3v2.service
 └─763 /home/pi/OctoPrint/bin/python3 /home/pi/OctoPrint/bin/octoprint serve --config=/home/pi/.Ender3v2/config.yaml --basedir=/home/pi/.Ender3v2 --port=5000

May 03 21:15:00 raspberrypi octoprint[763]: 2024-05-03 21:15:00,115 - octoprint.plugins.SpoolManager - INFO - Send Event 'plugin_spoolmanager_spool_selected' with payload '{'toolId': 0, 'databaseId': 1, 'spoolName': 'HATCHBOX PETG', 'material': 'PETG', 'colorName': 'black', 'remainingWeight': -1.6999999999999993}' to event-bus
May 03 21:15:00 raspberrypi octoprint[763]: 2024-05-03 21:15:00,120 - octoprint.plugins.SpoolManager - INFO - Send Event 'plugin_spoolmanager_spool_selected' with payload '{'toolId': 0, 'databaseId': 1, 'spoolName': 'HATCHBOX PETG', 'material': 'PETG', 'colorName': 'black', 'remainingWeight': -1.6999999999999993}' to event-bus
May 03 21:17:15 raspberrypi octoprint[763]: 2024-05-03 21:17:15,172 - octoprint.plugins.octoeverywhere - INFO - Webcam hardcode fallback http URL threw an exception: HTTPConnectionPool(host='127.0.0.1', port=8080): Max retries exceeded with url: /?action=snapshot (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fff1c6610d0>: Failed to establish a new connection: [Errno 111] Connection refused'))
May 03 21:17:15 raspberrypi octoprint[763]: 2024-05-03 21:17:15,172 - octoprint.plugins.octoeverywhere - ERROR - Webcam hardcode fallback failed and we have no more fallbacks. We DON'T have a main response.
May 03 21:17:15 raspberrypi octoprint[763]: 2024-05-03 21:17:15,190 - octoprint.plugins.octoeverywhere - INFO - Webcam hardcode fallback http URL threw an exception: HTTPConnectionPool(host='127.0.0.1', port=8080): Max retries exceeded with url: /?action=stream (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fff1c408950>: Failed to establish a new connection: [Errno 111] Connection refused'))
May 03 21:17:15 raspberrypi octoprint[763]: 2024-05-03 21:17:15,190 - octoprint.plugins.octoeverywhere - ERROR - Webcam hardcode fallback failed and we have no more fallbacks. We DON'T have a main response.
May 03 21:17:15 raspberrypi octoprint[763]: 2024-05-03 21:17:15,190 - octoprint.plugins.octoeverywhere - WARNING - Web Stream http [2] failed to make http request. octoHttpResult was None; url:/webcam/?action=snapshot
May 03 21:19:40 raspberrypi octoprint[763]: 2024-05-03 21:19:40,869 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:10.8.2.166
May 03 21:21:51 raspberrypi octoprint[763]: 2024-05-03 21:21:51,539 - octoprint.server.heartbeat - INFO - Server heartbeat <3
May 03 21:21:57 raspberrypi octoprint[763]: 2024-05-03 21:21:57,092 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 27905, 'printer_state': 'OPERATIONAL'}
None
Initializing PyArcWelder V0.1.0rc1.dev2 - Copyright (C) 2019 Brad Hochgesang. Python 3+ Detected...Complete.
 Initialization Complete
None
Initializing PyArcWelder V0.1.0rc1.dev2 - Copyright (C) 2019 Brad Hochgesang. Python 3+ Detected...Complete.
 Initialization Complete
{'bitrate': '10000k',
 'cleanTmpAfterDays': 7,
 'defaultWebcam': 'classic',
 'ffmpeg': None,
 'ffmpegCommandline': '{ffmpeg} -framerate {fps} -i "{input}" -vcodec '
 '{videocodec} -threads {threads} -b:v {bitrate} -f '
 '{containerformat} -y {filters} "{output}"',
 'ffmpegThreads': 1,
 'ffmpegThumbnailCommandline': '{ffmpeg} -sseof -1 -i "{input}" -update 1 -q:v '
 '0.7 "{output}"',
[... 27 lines cut ...]
May 03 13:36:43 raspberrypi camera-streamer[910]: device/device.c: CAMERA: Setting frame interval_us=0 for FPS=0
May 03 13:36:43 raspberrypi camera-streamer[910]: device/buffer_list.c: CAMERA:capture: Streaming started... Was 0 of 3 enqueud
[... cut ...]
 └─1037 /usr/sbin/haproxy -Ws -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid -S /run/haproxy-master.sock

May 03 13:36:49 raspberrypi systemd[1]: Starting haproxy.service - HAProxy Load Balancer...
May 03 13:36:50 raspberrypi haproxy[976]: [NOTICE] (976) : New worker (1037) forked
May 03 13:36:50 raspberrypi haproxy[976]: [NOTICE] (976) : Loading success.
May 03 13:36:50 raspberrypi systemd[1]: Started haproxy.service - HAProxy Load Balancer.

*************************
octoprint_deploy 1.0.11
*************************

Blisk, 2024-05-04:
> You have to provide more information. Please re-run octoprint_deploy and use Utilties->Diagnostic Output.

I did this on raspberry pi 3 and it worked now on raspberry 4 doesn't work I started all over 3 times and it doesn't work. I will on Monday do again and will post here from start to end.

Blisk, 2024-05-07:
I have start this all over again and again second camera doesn't work. 
Here is diagnostic
octoprint_deploy diagnostic information. Please provide ALL output for support help
**************************************
/etc/octoprint_deploy
**************************************
octoexec: /home/damjan/oprint/bin/octoprint
octopip: /home/damjan/oprint/bin/pip
haproxy: true
octopi: true
streamer: camera-streamer
**************************************
/etc/octoprint_instances
**************************************
instance:octoprint port:5000 udev:true
instance:MAX10 port:5001 udev:true
instance:ENDER3 port:5002 udev:true
**************************************
/etc/octoprint_cameras
**************************************
camera:cam_MAX10 port:8001 udev:true
camera:cam_ENDER3 port:8002 udev:true
**************************************
/etc/udev/rules.d/99-octoprint.rules
**************************************
SUBSYSTEM=="tty", ATTRS{serial}=="AM00OVSF", SYMLINK+="octo_MAX10"
SUBSYSTEM=="video4linux", ATTRS{serial}=="5516DFCF", ATTR{index}=="0", SYMLINK+="cam_MAX10"
SUBSYSTEM=="tty", ATTRS{serial}=="20873155574B", SYMLINK+="octo_ENDER3"
SUBSYSTEM=="video4linux", ATTRS{serial}=="0D1A8FCF", ATTR{index}=="0", SYMLINK+="cam_ENDER3"
lrwxrwxrwx 1 root root 7 May  7 13:50 /dev/octo_ENDER3 -> ttyACM0
lrwxrwxrwx 1 root root 7 May  7 13:50 /dev/octo_MAX10 -> ttyUSB0
lrwxrwxrwx 1 root root 6 May  7 13:50 /dev/cam_ENDER3 -> video2
lrwxrwxrwx 1 root root 6 May  7 13:50 /dev/cam_MAX10 -> video0
**************************************
● octoprint.service - The snappy web interface for your 3D printer
     Loaded: loaded (/etc/systemd/system/octoprint.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2024-05-07 13:50:35 CEST; 5min ago
   Main PID: 427 (octoprint)
      Tasks: 16 (limit: 4915)
        CPU: 19.800s
     CGroup: /system.slice/octoprint.service
             └─427 /home/damjan/oprint/bin/python /home/damjan/oprint/bin/octoprint serve --host=127.0.0.1 --port=5000

May 07 13:55:02 pi4 octoprint[427]: 2024-05-07 13:55:02,440 - octoprint.server.util.sockjs - INFO - User damjan logged in on the socket from client ::ffff:192.168.0.138
May 07 13:55:02 pi4 octoprint[427]: 2024-05-07 13:55:02,492 - octoprint.server.util.flask - INFO - Passively logging in user damjan from ::ffff:192.168.0.138
May 07 13:55:02 pi4 octoprint[427]: 2024-05-07 13:55:02,492 - octoprint.access.users - INFO - Logged in user: damjan
May 07 13:55:02 pi4 octoprint[427]: 2024-05-07 13:55:02,544 - octoprint.plugins.pluginmanager - INFO - Looks like we are offline, can't fetch notices from network
May 07 13:55:14 pi4 octoprint[427]: 2024-05-07 13:55:14,864 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.0.138
May 07 13:55:16 pi4 octoprint[427]: 2024-05-07 13:55:16,034 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.0.138
May 07 13:55:16 pi4 octoprint[427]: 2024-05-07 13:55:16,148 - octoprint.server.util.flask - INFO - Passively logging in user damjan from ::ffff:192.168.0.138
May 07 13:55:16 pi4 octoprint[427]: 2024-05-07 13:55:16,148 - octoprint.access.users - INFO - Logged in user: damjan
May 07 13:55:29 pi4 octoprint[427]: 2024-05-07 13:55:29,543 - octoprint.plugins.announcements - INFO - Looks like we are offline, can't fetch announcements for channel _important from network
May 07 13:55:41 pi4 octoprint[427]: 2024-05-07 13:55:41,713 - octoprint.plugins.announcements - INFO - Looks like we are offline, can't fetch announcements for channel _releases from network
None
None
{'bitrate': '10000k',
 'cleanTmpAfterDays': 7,
 'defaultWebcam': 'classic',
 'ffmpeg': None,
 'ffmpegCommandline': '{ffmpeg} -framerate {fps} -i "{input}" -vcodec '
                      '{videocodec} -threads {threads} -b:v {bitrate} -f '
                      '{containerformat} -y {filters} "{output}"',
 'ffmpegThreads': 1,
 'ffmpegThumbnailCommandline': '{ffmpeg} -sseof -1 -i "{input}" -update 1 -q:v '
                               '0.7 "{output}"',
 'ffmpegVideoCodec': 'libx264',
 'snapshotWebcam': 'classic',
 'timelapse': {'fps': 25,
               'options': {'capturePostRoll': None,
                           'interval': None,
[... 105 lines cut ...]
        CPU: 2ms
**************************************
● haproxy.service - HAProxy Load Balancer
     Loaded: loaded (/lib/systemd/system/haproxy.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2024-05-07 13:52:45 CEST; 3min 10s ago
       Docs: man:haproxy(1)
             file:/usr/share/doc/haproxy/configuration.txt.gz
    Process: 1115 ExecStartPre=/usr/sbin/haproxy -f $CONFIG -c -q $EXTRAOPTS (code=exited, status=0/SUCCESS)
   Main PID: 1123 (haproxy)
      Tasks: 5 (limit: 4915)
        CPU: 231ms
     CGroup: /system.slice/haproxy.service
             ├─1123 /usr/sbin/haproxy -Ws -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid -S /run/haproxy-master.sock
             └─1131 /usr/sbin/haproxy -Ws -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid -S /run/haproxy-master.sock

May 07 13:55:16 pi4 haproxy[1131]: ::ffff:192.168.0.138:64835 [07/May/2024:13:55:16.295] public octoprint/octoprint1 3/0/0/446/449 200 20222 - - ---- 5/5/1/1/0 0/0 "GET /api/settings HTTP/1.1"
May 07 13:55:16 pi4 haproxy[1131]: ::ffff:192.168.0.138:64835 [07/May/2024:13:55:16.824] public octoprint/octoprint1 0/0/1/6/7 304 357 - - ---- 5/5/1/1/0 0/0 "GET /static/img/mask-theme.svg HTTP/1.1"
[... cut ...]
May 07 13:55:17 pi4 haproxy[1131]: ::ffff:192.168.0.138:65124 [07/May/2024:13:55:17.276] public octoprint/octoprint1 2/0/1/55/58 200 6158 - - ---- 8/8/4/4/0 0/0 "GET /plugin/logging/ HTTP/1.1"
May 07 13:55:17 pi4 haproxy[1131]: ::ffff:192.168.0.138:65125 [07/May/2024:13:55:17.291] public octoprint/octoprint1 2/0/0/60/62 200 957 - - ---- 8/8/3/3/0 0/0 "GET /api/timelapse?unrendered=true HTTP/1.1"
May 07 13:55:17 pi4 haproxy[1131]: ::ffff:192.168.0.138:64835 [07/May/2024:13:55:17.318] public octoprint/octoprint1 0/0/0/45/45 200 801 - - ---- 8/8/4/4/0 0/0 "GET /api/plugin/action_command_notification HTTP/1.1"

paukstelis, 2024-05-07:
I would recommend NOT using OctoPi if you want to use octoprint_deploy to setup cameras. As the README states, OctoPi has its own way of doing cameras. Just install RPiOS and use octoprint_deploy that way.

Blisk, 2024-05-07:
> I would recommend NOT using OctoPi if you want to use octoprint_deploy to setup cameras. As the README states, OctoPi has its own way of doing cameras. Just install RPiOS and use octoprint_deploy that way.

thank you. I will install pure RPiOS I assume the one with pure debian and no desktop.

Blisk, 2024-05-08:
I did as you said and install pure RPOS and deployment and now none of camera works.
here is diagnostic output.

Select an option: 8
octoprint_deploy diagnostic information. Please provide ALL output for support help
**************************************
/etc/octoprint_deploy
**************************************
haproxy: true
streamer: ustreamer
octoexec: /home/damjan/OctoPrint/bin/octoprint
octopip: /home/damjan/OctoPrint/bin/pip
**************************************
/etc/octoprint_instances
**************************************
instance:MAX10 port:5000 udev:true
instance:ENDER port:5001 udev:true
**************************************
/etc/octoprint_cameras
**************************************
camera:cam_MAX10 port:8001 udev:true
camera:cam_ENDER port:8002 udev:true
**************************************
/etc/udev/rules.d/99-octoprint.rules
**************************************
SUBSYSTEM=="tty", ATTRS{serial}=="AM00OVSF", SYMLINK+="octo_MAX10"
SUBSYSTEM=="video4linux", ATTRS{serial}=="5516DFCF", ATTR{index}=="0", SYMLINK+="cam_MAX10"
SUBSYSTEM=="tty", ATTRS{serial}=="20873155574B", SYMLINK+="octo_ENDER"
SUBSYSTEM=="video4linux", ATTRS{serial}=="0D1A8FCF", ATTR{index}=="0", SYMLINK+="cam_ENDER"
lrwxrwxrwx 1 root root 7 May  8 09:05 /dev/octo_ENDER -> ttyACM0
lrwxrwxrwx 1 root root 7 May  8 09:05 /dev/octo_MAX10 -> ttyUSB0
lrwxrwxrwx 1 root root 6 May  8 09:05 /dev/cam_ENDER -> video2
lrwxrwxrwx 1 root root 6 May  8 09:05 /dev/cam_MAX10 -> video0
**************************************
● MAX10.service - The snappy web interface for your 3D printer
     Loaded: loaded (/etc/systemd/system/MAX10.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2024-05-08 09:05:42 CEST; 2min 22s ago
   Main PID: 368 (octoprint)
      Tasks: 28 (limit: 3933)
        CPU: 25.590s
     CGroup: /system.slice/MAX10.service
             └─368 /home/damjan/OctoPrint/bin/python3 /home/damjan/OctoPrint/bin/octoprint serve --config=/home/damjan/.MAX10/config.yaml --basedir=/home/damjan/.MAX10 --port=5000

May 08 09:06:57 pi4 octoprint[368]: 2024-05-08 09:06:57,509 - octoprint.plugins.file_check - INFO - ... got 0 matches
May 08 09:06:57 pi4 octoprint[368]: 2024-05-08 09:06:57,510 - octoprint.plugins.file_check - INFO - Running check leaked_api_key
May 08 09:06:57 pi4 octoprint[368]: 2024-05-08 09:06:57,535 - octoprint.plugins.file_check - INFO - ... got 0 matches
May 08 09:07:25 pi4 octoprint[368]: 2024-05-08 09:07:25,321 - octoprint.server.util.flask - INFO - Passively logging in user damjan from ::ffff:192.168.0.138
May 08 09:07:25 pi4 octoprint[368]: 2024-05-08 09:07:25,322 - octoprint.access.users - INFO - Logged in user: damjan
May 08 09:07:25 pi4 octoprint[368]: 2024-05-08 09:07:25,859 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.0.138
May 08 09:07:26 pi4 octoprint[368]: 2024-05-08 09:07:26,364 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.0.138
May 08 09:07:26 pi4 octoprint[368]: 2024-05-08 09:07:26,404 - octoprint.server.util.flask - INFO - Passively logging in user damjan from ::ffff:192.168.0.138
May 08 09:07:26 pi4 octoprint[368]: 2024-05-08 09:07:26,404 - octoprint.access.users - INFO - Logged in user: damjan
May 08 09:07:26 pi4 octoprint[368]: 2024-05-08 09:07:26,645 - octoprint.server.util.sockjs - INFO - User damjan logged in on the socket from client ::ffff:192.168.0.138
None
None
{'bitrate': '10000k',
 'cleanTmpAfterDays': 7,
 'defaultWebcam': 'classic',
 'ffmpeg': None,
 'ffmpegCommandline': '{ffmpeg} -framerate {fps} -i "{input}" -vcodec '
                      '{videocodec} -threads {threads} -b:v {bitrate} -f '
                      '{containerformat} -y {filters} "{output}"',
 'ffmpegThreads': 1,
 'ffmpegThumbnailCommandline': '{ffmpeg} -sseof -1 -i "{input}" -update 1 -q:v '
                               '0.7 "{output}"',
 'ffmpegVideoCodec': 'libx264',
 'snapshotWebcam': 'classic',
 'timelapse': {'fps': 25,
               'options': {'capturePostRoll': None,
                           'interval': None,
[... 82 lines cut ...]
May 08 09:08:19 pi4 ustreamer[370]: -- ERROR [151.448          ] -- CAP: Can't consume V4L2 event: No such device
May 08 09:08:19 pi4 ustreamer[370]: -- INFO  [151.458          ] -- Destroying workers pool JPEG ...
May 08 09:08:19 pi4 ustreamer[370]: -- ERROR [151.458          ] -- CAP: Can't stop capturing: No such device
May 08 09:08:19 pi4 ustreamer[370]: -- INFO  [151.460          ] -- CAP: Capturing stopped
May 08 09:08:19 pi4 ustreamer[370]: ================================================================================
May 08 09:08:19 pi4 ustreamer[370]: -- ERROR [151.460          ] -- No access to capture device: No such file or directory
May 08 09:08:19 pi4 ustreamer[370]: -- INFO  [151.460          ] -- Waiting for the capture device ...
**************************************
● haproxy.service - HAProxy Load Balancer
     Loaded: loaded (/lib/systemd/system/haproxy.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2024-05-08 09:05:44 CEST; 2min 37s ago
       Docs: man:haproxy(1)
             file:/usr/share/doc/haproxy/configuration.txt.gz
    Process: 564 ExecStartPre=/usr/sbin/haproxy -f $CONFIG -c -q $EXTRAOPTS (code=exited, status=0/SUCCESS)
   Main PID: 583 (haproxy)
      Tasks: 5 (limit: 3933)
        CPU: 445ms
     CGroup: /system.slice/haproxy.service
[... cut ...]
             └─594 /usr/sbin/haproxy -Ws -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid -S /run/haproxy-master.sock

May 08 09:05:44 pi4 systemd[1]: Starting HAProxy Load Balancer...
May 08 09:05:44 pi4 haproxy[583]: [WARNING] 128/090544 (583) : parsing [/etc/haproxy/haproxy.cfg:39] : a 'http-request' rule placed after a 'use_backend' rule will still be processed before.
May 08 09:05:44 pi4 haproxy[583]: [NOTICE] 128/090544 (583) : New worker #1 (594) forked
May 08 09:05:44 pi4 systemd[1]: Started HAProxy Load Balancer.

paukstelis, 2024-05-08:
I don't know why it isn't working in this case. Clearly, the cameras are present and the udev symlinks are there. The correct device is listed in ustreamer.

You could try deleting the cameras, changing the streamer to camera-streamer, adding the cameras back, and seeing if that works.

Blisk, 2024-05-13:
I found out there were no settings in stream URL I filled that up and now it works

Efekent, 2024-08-10:
> I found out there were no settings in stream URL I filled that up and now it works

Hello @Blisk , I have the same issue, both cameras show up in the Diagnostic Output, but I can't find the right URL for it:
`camera:cam_4Max port:8001 udev:true
camera:cam_TevoMA port:8002 udev:true`

`lrwxrwxrwx 1 root root 6 Aug 10 14:01 /dev/cam_4Max -> video0
lrwxrwxrwx 1 root root 6 Aug 10 14:01 /dev/cam_TevoMA -> video2`

Could you please show me what URL you have used as example?

Thanks.

Blisk, 2024-08-26:
> EDIT: I got it working after reading another post, deleted both cameras and changed streamer to ustreamer and added cameras back, now both cameras work automatically, no need for URL.
> 
> ############## Original question #############
> 
> > I found out there were no settings in stream URL I filled that up and now it works
> 
> Hello @Blisk , I have the same issue, both cameras show up in the Diagnostic Output, but I can't find the right URL for it: Could you please show me what URL you have used as example?
> 
> Thanks.

[image]