paukstelis/octoprint_deploy #165: Raspberry pi 5 usb can not working

ingo916, 2024-05-05:
Used the octoprint_deploy and installed octoprint. It works fine; however, the USB camera is not working.
Installed the camera using the methods described in the github page.

Log output below
*************************
octoprint_deploy 1.0.11
*************************

1) Add instance     4) Add PiCam        7) Backup Menu
2) Delete instance  5) Delete Camera    8) Update
3) Add USB Camera   6) Utilities        9) Quit
Select operation: 
1) Add instance     4) Add PiCam        7) Backup Menu
2) Delete instance  5) Delete Camera    8) Update
3) Add USB Camera   6) Utilities        9) Quit
Select operation: 
1) Add instance     4) Add PiCam        7) Backup Menu
2) Delete instance  5) Delete Camera    8) Update
3) Add USB Camera   6) Utilities        9) Quit
Select operation: 6

1) Instance Status    4) Share Uploads      7) Udev Menu
2) USB Port Testing   5) Change Streamer    8) Diagnostic Output
3) Sync Users         6) Set Global Config  9) Quit
Select an option: 8
octoprint_deploy diagnostic information. Please provide ALL output for support help
**************************************
/etc/octoprint_deploy
**************************************
haproxy: true
octoexec: /home/pi/OctoPrint/bin/octoprint
octopip: /home/pi/OctoPrint/bin/pip
streamer: ustreamer
**************************************
/etc/octoprint_instances
**************************************
instance:Ender3v2 port:5000 udev:true
**************************************
/etc/octoprint_cameras
**************************************
camera:cam_Ender3v2 port:8001 udev:true
**************************************
/etc/udev/rules.d/99-octoprint.rules
**************************************
KERNELS=="3-1:1.0",SUBSYSTEM=="tty",SYMLINK+="octo_Ender3v2"
SUBSYSTEM=="video4linux", ATTRS{serial}=="43E9B72F", ATTR{index}=="0", SYMLINK+="cam_Ender3v2"
lrwxrwxrwx 1 root root 7 May  5 12:22 /dev/octo_Ender3v2 -> ttyUSB0
lrwxrwxrwx 1 root root 6 May  5 12:22 /dev/cam_Ender3v2 -> video0
**************************************
â— Ender3v2.service - The snappy web interface for your 3D printer
     Loaded: loaded (/etc/systemd/system/Ender3v2.service; enabled; preset: enabled)
     Active: active (running) since Sun 2024-05-05 12:22:24 PDT; 1min 18s ago
   Main PID: 1861 (octoprint)
      Tasks: 42 (limit: 9247)
        CPU: 12.669s
     CGroup: /system.slice/Ender3v2.service
             â””â”€1861 /home/pi/OctoPrint/bin/python3 /home/pi/OctoPrint/bin/octoprint serve --config=/home/pi/.Ender3v2/config.yaml --basedir=/home/pi/.Ender3v2 --port=5000

May 05 12:23:07 raspberrypi octoprint[1861]: 2024-05-05 12:23:07,215 - octoprint.plugins.SpoolManager - INFO - Send Event 'plugin_spoolmanager_spool_selected' with payload '{'toolId': 0, 'databaseId': 1, 'spoolName': 'HATCHBOX PETG', 'material': 'PETG', 'colorName': 'black', 'remainingWeight': -1.6999999999999993}' to event-bus
May 05 12:23:07 raspberrypi octoprint[1861]: 2024-05-05 12:23:07,278 - octoprint.plugins.SpoolManager - INFO - Send Event 'plugin_spoolmanager_spool_selected' with payload '{'toolId': 0, 'databaseId': 1, 'spoolName': 'HATCHBOX PETG', 'material': 'PETG', 'colorName': 'black', 'remainingWeight': -1.6999999999999993}' to event-bus
May 05 12:23:07 raspberrypi octoprint[1861]: 2024-05-05 12:23:07,281 - octoprint.plugins.SpoolManager - INFO - Send Event 'plugin_spoolmanager_spool_selected' with payload '{'toolId': 0, 'databaseId': 1, 'spoolName': 'HATCHBOX PETG', 'material': 'PETG', 'colorName': 'black', 'remainingWeight': -1.6999999999999993}' to event-bus
May 05 12:23:07 raspberrypi octoprint[1861]: 2024-05-05 12:23:07,286 - octoprint.plugins.SpoolManager - INFO - Send Event 'plugin_spoolmanager_spool_selected' with payload '{'toolId': 0, 'databaseId': 1, 'spoolName': 'HATCHBOX PETG', 'material': 'PETG', 'colorName': 'black', 'remainingWeight': -1.6999999999999993}' to event-bus
May 05 12:23:07 raspberrypi octoprint[1861]: 2024-05-05 12:23:07,292 - octoprint.plugins.SpoolManager - INFO - Send Event 'plugin_spoolmanager_spool_selected' with payload '{'toolId': 0, 'databaseId': 1, 'spoolName': 'HATCHBOX PETG', 'material': 'PETG', 'colorName': 'black', 'remainingWeight': -1.6999999999999993}' to event-bus
May 05 12:23:07 raspberrypi octoprint[1861]: 2024-05-05 12:23:07,348 - octoprint.server.util.sockjs - INFO - User ingo logged in on the socket from client ::ffff:10.8.2.49
May 05 12:23:08 raspberrypi octoprint[1861]: 2024-05-05 12:23:08,047 - octoprint.plugins.tracking - INFO - Sent tracking event webui_load, payload: {'browser_name': 'Chrome', 'browser_version': '124.0.0.0', 'os_name': 'Windows', 'os_version': '10'}
May 05 12:23:08 raspberrypi octoprint[1861]: 2024-05-05 12:23:08,733 - octoprint.plugins.SpoolManager - INFO - Send Event 'plugin_spoolmanager_spool_selected' with payload '{'toolId': 0, 'databaseId': 1, 'spoolName': 'HATCHBOX PETG', 'material': 'PETG', 'colorName': 'black', 'remainingWeight': -1.6999999999999993}' to event-bus
May 05 12:23:08 raspberrypi octoprint[1861]: 2024-05-05 12:23:08,860 - octoprint.plugins.SpoolManager - INFO - Send Event 'plugin_spoolmanager_spool_selected' with payload '{'toolId': 0, 'databaseId': 1, 'spoolName': 'HATCHBOX PETG', 'material': 'PETG', 'colorName': 'black', 'remainingWeight': -1.6999999999999993}' to event-bus
May 05 12:23:08 raspberrypi octoprint[1861]: 2024-05-05 12:23:08,863 - octoprint.plugins.SpoolManager - INFO - Send Event 'plugin_spoolmanager_spool_selected' with payload '{'toolId': 0, 'databaseId': 1, 'spoolName': 'HATCHBOX PETG', 'material': 'PETG', 'colorName': 'black', 'remainingWeight': -1.6999999999999993}' to event-bus
None
Initializing PyArcWelder V0.1.0rc1.dev2 - Copyright (C) 2019  Brad Hochgesang. Python 3+ Detected...Complete.
[... 41 lines cut ...]
May 05 12:22:32 raspberrypi ustreamer[1882]: -- INFO  [830.072          ] -- Switching to HW encoder: the input is (M)JPEG ...
May 05 12:22:32 raspberrypi ustreamer[1882]: -- INFO  [830.072          ] -- Using JPEG quality: encoder default
May 05 12:22:32 raspberrypi ustreamer[1882]: -- INFO  [830.072          ] -- Creating pool JPEG with 1 workers ...
May 05 12:22:32 raspberrypi ustreamer[1882]: -- INFO  [830.072          ] -- Capturing ...
**************************************
â— haproxy.service - HAProxy Load Balancer
     Loaded: loaded (/lib/systemd/system/haproxy.service; enabled; preset: enabled)
     Active: active (running) since Sun 2024-05-05 12:22:24 PDT; 1min 22s ago
       Docs: man:haproxy(1)
             file:/usr/share/doc/haproxy/configuration.txt.gz
   Main PID: 1873 (haproxy)
      Tasks: 5 (limit: 9247)
        CPU: 101ms
     CGroup: /system.slice/haproxy.service
             â”œâ”€1873 /usr/sbin/haproxy -Ws -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid -S /run/haproxy-master.sock
             â””â”€1875 /usr/sbin/haproxy -Ws -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid -S /run/haproxy-master.sock

May 05 12:22:24 raspberrypi systemd[1]: Starting haproxy.service - HAProxy Load Balancer...
May 05 12:22:24 raspberrypi haproxy[1873]: [NOTICE]   (1873) : New worker (1875) forked
May 05 12:22:24 raspberrypi haproxy[1873]: [NOTICE]   (1873) : Loading success.
May 05 12:22:24 raspberrypi systemd[1]: Started haproxy.service - HAProxy Load Balancer.

*************************
octoprint_deploy 1.0.11
*************************

borkedporcupine, 2025-04-08:
Same here, the script is able to detect and add my camera, but the camera isn't seen in octoprint.  Using ustreamer as well, was going to re-flash and try the other options to see if they end up working.  Be happy to see anyone else able to get this working.

Dawied, 2025-05-18:
Same here, USB camera is working on standard /webcam/?action=stream, but after adding the same camera with octoprint_deplay and haproxy the stream is not working (cam_mk3s/?action=stream).