paukstelis/octoprint_deploy #167: How to add integrated laptop cam

OscarCreator, 2024-05-11:
I got octoprint working with this script on my old laptop but couldn't find any way to configure the integrated laptop cam as the webcam. How would I do that?

paukstelis, 2024-05-13:
There isn't exactly an "easy" way to do this, because the script requires that you plug the camera in. One possible solution is to plug in some other USB device that has a serial number. Then you can modify either the service file or the udev rules to point to your integrated camera (probably /dev/video0)

OscarCreator, 2024-05-13:
Okay, thanks for the help. I will try and see if I can get this to work.

OscarCreator, 2024-05-14:
Wow, picam and the editing the device to /dev/video0 totally worked! 🥳
[image]

closed 2024-05-14 by OscarCreator