paukstelis/octoprint_deploy #171: Can't install Octolapse or Arc Welder

thedasfaust, 2024-06-24:
This is the output from trying to install Octolapse, but I get the same results with Arc Welder, is this a problem with the plugins or my installation of Octoprint on Ubuntu?

Installing plugin "Octolapse" from https://github.com/FormerLurker/Octolapse/archive/master.zip...
/home/das/OctoPrint/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpb2p2umyz/Octolapse-master.zip --no-cache-dir
Processing /tmp/tmpb2p2umyz/Octolapse-master.zip
Preparing metadata (setup.py): started
Preparing metadata (setup.py): finished with status 'error'
error: subprocess-exited-with-error

× python setup.py egg_info did not run successfully.
│ exit code: 1
╰─> [15 lines of output]
/tmp/pip-req-build-ncy31_kz/versioneer.py:421: SyntaxWarning: invalid escape sequence '\s'
LONG_VERSION_PY['git'] = '''
Traceback (most recent call last):
File "<string>", line 2, in <module>
File "<pip-setuptools-caller>", line 34, in <module>
File "/tmp/pip-req-build-ncy31_kz/setup.py", line 25, in <module>
plugin_version = NumberedVersion.clean_version(versioneer.get_versions(verbose=True)["version"])
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/tmp/pip-req-build-ncy31_kz/versioneer.py", line 1412, in get_versions
cfg = get_config_from_root(root)
^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/tmp/pip-req-build-ncy31_kz/versioneer.py", line 342, in get_config_from_root
parser = configparser.SafeConfigParser()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'configparser' has no attribute 'SafeConfigParser'. Did you mean: 'RawConfigParser'?
[end of output]

note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

paukstelis, 2024-06-24:
This isn't an octoprint_deploy problem. This is almost certainly a Python version issue. I'm guessing you are on Python3.12 which will give the regex escape character error.

thedasfaust, 2024-06-24:
So I should figure out how to remove python 3.12.3 and install an older version?

thedasfaust, 2024-06-24:
On the latest version of Ubuntu LTS desktop

paukstelis, 2024-06-24:
Later today I will push a change for octoprint_deploy that will pin python version to 3.11. I will let you know here once it is done. You'll want to update octoprint_deploy, run it with the `remove` argument, then start again.

thedasfaust, 2024-06-24:
I've removed and ran the ocotpirnt_deploy script several times already, that's the easy part. I've reinstalled Ubuntu twice so far because of the rabbit hole I went down trying to "fix" things. I'm a little bit of a noob at LiNUX so this just helps me out in the long run actually. Thank you

paukstelis, 2024-06-24:
Could you please do the following and report back if it works?
`git -C octoprint_deploy pull`
`git -C octoprint_deploy checkout python3.11`
`sudo octoprint_deploy/octoprint_deploy.sh remove`
Answer Y, then do the setup process. If you can save that output it would be useful to paste here.

thedasfaust, 2024-06-24:
As soon as I finish up a print I will do that. Thank you

thedasfaust, 2024-06-24:
Installation type: 1
Ready to begin? [y/n]: y
Adding current user to dialout and video groups.
Adding systemctl and reboot to sudo
This will install necessary packages, install OctoPrint and setup an
instance.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
virtualenv is already the newest version (20.25.0+ds-2).
build-essential is already the newest version (12.10ubuntu1).
g++ is already the newest version (4:13.2.0-7ubuntu1).
gcc is already the newest version (4:13.2.0-7ubuntu1).
libjpeg-dev is already the newest version (8c-2ubuntu11).
ssh is already the newest version (1:9.6p1-3ubuntu13).
python3-venv is already the newest version (3.12.3-0ubuntu1).
python-is-python3 is already the newest version (3.11.4-1).
xxd is already the newest version (2:9.1.0016-1ubuntu7.1).
libcamera-v4l2 is already the newest version (0.2.0-3fakesync1build6).
ffmpeg is already the newest version (7:6.1.1-3ubuntu5).
libcamera-dev is already the newest version (0.2.0-3fakesync1build6).
libbsd-dev is already the newest version (0.12.1-1build1).
v4l-utils is already the newest version (1.26.1-4build3).
libffi-dev is already the newest version (3.4.6-1build1).
libcamera-tools is already the newest version (0.2.0-3fakesync1build6).
make is already the newest version (4.3-4.1build2).
uuid-runtime is already the newest version (2.39.3-9ubuntu6).
python3-dev is already the newest version (3.12.3-0ubuntu1).
libssl-dev is already the newest version (3.0.13-0ubuntu3.1).
rsync is already the newest version (3.2.7-1ubuntu1).
pkg-config is already the newest version (1.8.1-2build1).
python3-pip is already the newest version (24.0+dfsg-1ubuntu1).
libavutil-dev is already the newest version (7:6.1.1-3ubuntu5).
libyaml-dev is already the newest version (0.2.5-1build1).
haproxy is already the newest version (2.8.5-1ubuntu3).
python3-setuptools is already the newest version (68.1.2-2ubuntu1).
libavformat-dev is already the newest version (7:6.1.1-3ubuntu5).
libavcodec-dev is already the newest version (7:6.1.1-3ubuntu5).
libjpeg8-dev is already the newest version (8c-2ubuntu11).
cmake is already the newest version (3.28.3-1build7).
libevent-dev is already the newest version (2.1.12-stable-9ubuntu2).
0 upgraded, 0 newly installed, 0 to remove and 11 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package 'brltty' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 11 not upgraded.
Enabling ssh server...
Synchronizing state of ssh.service with SysV service script with
/usr/lib/systemd/systemd-sysv-install.
Executing: /usr/lib/systemd/systemd-sysv-install enable ssh
Installing OctoPrint virtual environment in /home/das/OctoPrint
sudo: python3.11: command not found
sudo: /home/das/OctoPrint/bin/pip: command not found
sudo: /home/das/OctoPrint/bin/pip: command not found
sudo: /home/das/OctoPrint/bin/pip: command not found
WARNING! WARNING! WARNING!
OctoPrint has not been installed correctly.
Please answer Y to remove everything and try running prepare system again.
Remove everything? [y/n]:

It kind of just loops like this

paukstelis, 2024-06-24:
try `sudo apt install pytho3.11-dev` and then try again

thedasfaust, 2024-06-24:
added a repo, then updated and it installed

thedasfaust, 2024-06-24:
Building wheels for collected packages: netifaces
  Building wheel for netifaces (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [22 lines of output]

/home/das/OctoPrint/lib/python3.11/site-packages/setuptools/config/setupcfg.py:508:
SetuptoolsDeprecationWarning: The license_file parameter is deprecated, use
license_files instead.
        warnings.warn(msg, warning_class)
      running bdist_wheel
      running build
      running build_ext
      checking for getifaddrs...found.
      checking for getnameinfo...found.
      checking for IPv6 socket IOCTLs...not found.
      checking for optional header files...netash/ash.h netatalk/at.h
netax25/ax25.h neteconet/ec.h netipx/ipx.h netpacket/packet.h
netrose/rose.h linux/atm.h linux/llc.h linux/tipc.h.
      checking whether struct sockaddr has a length field...no.
      checking which sockaddr_xxx structs are defined...at ax25 in in6 ipx
un rose ash ec ll atmpvc atmsvc llc.
      checking for routing socket support...no.
      checking for sysctl(CTL_NET...) support...no.
      checking for netlink support...yes.
      will use netlink to read routing table
      building 'netifaces' extension
      x86_64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g
-fno-omit-frame-pointer -mno-omit-leaf-frame-pointer
-fstack-protector-strong -fstack-clash-protection -Wformat
-Werror=format-security -fcf-protection -g -fwrapv -O2 -fPIC
-DNETIFACES_VERSION=0.11.0 -DHAVE_GETIFADDRS=1 -DHAVE_GETNAMEINFO=1
-DHAVE_NETASH_ASH_H=1 -DHAVE_NETATALK_AT_H=1 -DHAVE_NETAX25_AX25_H=1
-DHAVE_NETECONET_EC_H=1 -DHAVE_NETIPX_IPX_H=1 -DHAVE_NETPACKET_PACKET_H=1
-DHAVE_NETROSE_ROSE_H=1 -DHAVE_LINUX_ATM_H=1 -DHAVE_LINUX_LLC_H=1
-DHAVE_LINUX_TIPC_H=1 -DHAVE_SOCKADDR_AT=1 -DHAVE_SOCKADDR_AX25=1
-DHAVE_SOCKADDR_IN=1 -DHAVE_SOCKADDR_IN6=1 -DHAVE_SOCKADDR_IPX=1
-DHAVE_SOCKADDR_UN=1 -DHAVE_SOCKADDR_ROSE=1 -DHAVE_SOCKADDR_ASH=1
-DHAVE_SOCKADDR_EC=1 -DHAVE_SOCKADDR_LL=1 -DHAVE_SOCKADDR_ATMPVC=1
-DHAVE_SOCKADDR_ATMSVC=1 -DHAVE_SOCKADDR_LLC=1 -DHAVE_PF_NETLINK=1
-I/home/das/OctoPrint/include -I/usr/include/python3.11 -c netifaces.c -o
build/temp.linux-x86_64-cpython-311/netifaces.o
      netifaces.c:1:10: fatal error: Python.h: No such file or directory
          1 | #include <Python.h>
            |          ^~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/x86_64-linux-gnu-gcc' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a
problem with pip.
  ERROR: Failed building wheel for netifaces
  Running setup.py clean for netifaces
Failed to build netifaces
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml
based projects (netifaces)
WARNING! WARNING! WARNING!
OctoPrint has not been installed correctly.
Please answer Y to remove everything and try running prepare system again.
Remove everything? [y/n]:

thedasfaust, 2024-06-24:
renamed the original directory, and got a fresh copy from github got a bit
further

paukstelis, 2024-06-24:
I think this is because there is cached things from your previous pip runs
you can try `rm -rf ~/.cache/pip` before installation.

thedasfaust, 2024-06-24:
Still failing on netifaces

deleted octoprint_deploy directory
recloned it
did the part pointing at python3.11
deleted the pip cache
ran the script

thedasfaust, 2024-06-25:
> You might try from a clean installation of the OS.

Did that last night, going to set up SSH and go from there this morning.

thedasfaust, 2024-06-25:
Could doing a pull (git -C octoprint_deploy pull) and having "Already up to
date" come back have anything to do with the latest issue I've encountered?

thedasfaust, 2024-06-25:
Fresh installs all around fixed it, thank you.

paukstelis, 2024-06-26:
Temporary fix with https://github.com/paukstelis/octoprint_deploy/pull/172. Closing.

closed 2024-06-26 by paukstelis