paukstelis/octoprint_deploy #173: Warning! OctoPrint has not been installed correctly.

pskouson, 2024-07-03:
Adding current user to dialout and video groups.
Adding systemctl and reboot to sudo
This will install necessary packages, install OctoPrint and setup an instance.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
virtualenv is already the newest version (20.25.0+ds-2).
build-essential is already the newest version (12.10ubuntu1).
g++ is already the newest version (4:13.2.0-7ubuntu1).
gcc is already the newest version (4:13.2.0-7ubuntu1).
libjpeg-dev is already the newest version (8c-2ubuntu11).
ssh is already the newest version (1:9.6p1-3ubuntu13.3).
python3-venv is already the newest version (3.12.3-0ubuntu1).
python-is-python3 is already the newest version (3.11.4-1).
xxd is already the newest version (2:9.1.0016-1ubuntu7.1).
xxd set to manually installed.
libcamera-v4l2 is already the newest version (0.2.0-3fakesync1build6).
ffmpeg is already the newest version (7:6.1.1-3ubuntu5).
libcamera-dev is already the newest version (0.2.0-3fakesync1build6).
libbsd-dev is already the newest version (0.12.1-1build1).
v4l-utils is already the newest version (1.26.1-4build3).
libffi-dev is already the newest version (3.4.6-1build1).
libcamera-tools is already the newest version (0.2.0-3fakesync1build6).
make is already the newest version (4.3-4.1build2).
uuid-runtime is already the newest version (2.39.3-9ubuntu6).
python3-dev is already the newest version (3.12.3-0ubuntu1).
libssl-dev is already the newest version (3.0.13-0ubuntu3.1).
rsync is already the newest version (3.2.7-1ubuntu1).
pkg-config is already the newest version (1.8.1-2build1).
python3-pip is already the newest version (24.0+dfsg-1ubuntu1).
libavutil-dev is already the newest version (7:6.1.1-3ubuntu5).
libyaml-dev is already the newest version (0.2.5-1build1).
haproxy is already the newest version (2.8.5-1ubuntu3).
python3-setuptools is already the newest version (68.1.2-2ubuntu1).
libavformat-dev is already the newest version (7:6.1.1-3ubuntu5).
libavcodec-dev is already the newest version (7:6.1.1-3ubuntu5).
libjpeg8-dev is already the newest version (8c-2ubuntu11).
cmake is already the newest version (3.28.3-1build7).
libevent-dev is already the newest version (2.1.12-stable-9ubuntu2).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package 'brltty' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Enabling ssh server...
Synchronizing state of ssh.service with SysV service script with /usr/lib/systemd/systemd-sysv-install.
Executing: /usr/lib/systemd/systemd-sysv-install enable ssh
Installing OctoPrint virtual environment in /home/howie/OctoPrint
sudo: python3.11: command not found
sudo: /home/xxxxx/OctoPrint/bin/pip: command not found
sudo: /home/xxxxx/OctoPrint/bin/pip: command not found
sudo: /home/xxxxx/OctoPrint/bin/pip: command not found
WARNING! WARNING! WARNING!
OctoPrint has not been installed correctly.
Please answer Y to remove everything and try running prepare system again.

I'm on a fresh install of Ubuntu Server.

paukstelis, 2024-07-03:
This is because the latest ubuntu (24) server does not have packages for older python versions. I pinned to python 3.11 because changes  in python3.12 breaks a lot of OctoPrint plugins. I suppose I will have no choice but to remove the pin. Keep in mind, many plugins won't work with python3.12.

paukstelis, 2024-07-03:
You can update octoprint_deploy directly form the octoprint_deploy menu and try again.

pskouson, 2024-07-03:
I guess in the meantime maybe I should download older ubuntu and try again maybe that'll be easiest for me. I've looked up how to downgrade python and screw that.

pskouson, 2024-07-11:
Okay I finally went back in to the school and ran update from the octoprint_deploy menu and I'm up and running. Thanks!