paukstelis/octoprint_deploy #174: Fix proxy. Updated script. Tried to remove and re-add pi cam. Broke everything.

skewll, 2024-08-28:
I was able to access my printers via,
 10.0.0.123/printer1
 10.0.0.123/printer2
 10.0.0.123/printer3
 10.0.0.123/printer4

Now I can only get to printer 1 via:

 10.0.0.123

 and cannot access other printers at all for some reason and I get the following screen when I try to visit the old links:

Not Found
The requested URL was not found on the server. If you entered the URL manually please check your spelling and try again.

How can I set the old mapping back up manually?

paukstelis, 2024-08-28:
Have you just done a reboot? You can try `sudo systemctl restart haproxy`

skewll, 2024-08-28:
Yes to reboot. I've restarted haproxy. Everything is still seems to be in
/etc/haproxy/haproxy.cfg

I can access the printers by their port via 10.0.0.124:5001 etc. But not by
name.

paukstelis, 2024-08-28:
That sounds like haproxy is not running. what do you get from `sudo systemctl status haproxy`?

skewll, 2024-08-28:
● haproxy.service - HAProxy Load Balancer
     Loaded: loaded (/lib/systemd/system/haproxy.service; enabled; preset:
enabled)
     Active: active (running) since Tue 2024-08-27 17:27:20 PDT; 15min ago
       Docs: man:haproxy(1)
             file:/usr/share/doc/haproxy/configuration.txt.gz
   Main PID: 801 (haproxy)
      Tasks: 5 (limit: 3832)
     Memory: 11.8M
        CPU: 534ms
     CGroup: /system.slice/haproxy.service
             ├─801 /usr/sbin/haproxy -Ws -f /etc/haproxy/haproxy.cfg -p
/run/haproxy.pid -S /run/haproxy-master>
             └─953 /usr/sbin/haproxy -Ws -f /etc/haproxy/haproxy.cfg -p
/run/haproxy.pid -S /run/haproxy-master>

Aug 27 17:26:48 printfarm systemd[1]: Starting haproxy.service - HAProxy
Load Balancer...
Aug 27 17:27:19 printfarm haproxy[801]: [NOTICE]   (801) : New worker (953)
forked
Aug 27 17:27:19 printfarm haproxy[801]: [NOTICE]   (801) : Loading success.
Aug 27 17:27:20 printfarm systemd[1]: Started haproxy.service - HAProxy
Load Balancer.
~
~
~
lines 1-17/17 (END)

paukstelis, 2024-08-28:
`sudo haproxy -f /etc/haproxy/haproxy.cfg -c` should tell you if there are errors  in haproxy config.

skewll, 2024-08-28:
It returns 'Configuration file is valid'

skewll, 2024-08-28:
One thing I did before it messed up, I was at the delete printer part of a
script, and has options 1,2,3 available.

1: cam_printer1
2: cam_printer2
3: quit

I pressed 9 my mistake because I was looking at a different output line,
but it still ran the remove script on a blank string. I think there was
some error output buy can't scroll back far enough confirm . I am curious
if that has some unexpected impact.

The only other thing I did prior was install raspi-config the enable the
picam. Never got pi cam to work. I tried removing raspi-config to see if
that fixed the url issue, but no luck.

paukstelis, 2024-08-29:
Can you  paste your haproxy.cfg here? It sounds like that is likely messed up by whatever happened.

skewll, 2024-08-29:
cat /etc/haproxy/haproxy.cfg
#Basic setup from:
#
https://community.octoprint.org/t/setting-up-octoprint-on-a-computer-running-fedora-centos-almalinux-or-rockylinux/37137/2

global
        maxconn 4000
        user haproxy
        group haproxy
        daemon
        log 127.0.0.1 local0 debug

defaults
        log     global
        mode    http
        option  httplog
        option  dontlognull
        retries 3
        option redispatch
        option http-server-close
        option forwardfor except 127.0.0.0/8
        timeout http-request    10s
        timeout queue           1m
        timeout connect         10s
        timeout client          1m
        timeout server          1m
        timeout http-keep-alive 10s
        timeout check           10s

frontend public
        bind :::80 v4v6
        option forwardfor except 127.0.0.1
#stella start
            acl is_stella url_beg /stella
            http-request redirect scheme http drop-query append-slash  if
is_stella ! { path_beg /stella/ }
            #stella stop
#dolly start
            acl is_dolly url_beg /dolly
            http-request redirect scheme http drop-query append-slash  if
is_dolly ! { path_beg /dolly/ }
            #dolly stop
#holly start
            acl is_holly url_beg /holly
            http-request redirect scheme http drop-query append-slash  if
is_holly ! { path_beg /holly/ }
            #holly stop
#molly start
            acl is_molly url_beg /molly
            http-request redirect scheme http drop-query append-slash  if
is_molly ! { path_beg /molly/ }
            #molly stop
        default_backend octoprint

backend octoprint
        option forwardfor
        server octoprint1 127.0.0.1:5000
#molly start
backend molly
       http-request replace-path /molly/(.*) /\1
       acl needs_scheme req.hdr_cnt(X-Scheme) eq 0
       http-request add-header X-Scheme https if needs_scheme { ssl_fc }
       http-request add-header X-Scheme http if needs_scheme !{ ssl_fc }
       http-request add-header X-Script-Name /molly
       server octoprint1 127.0.0.1:5000
       option forwardfor
#molly stop
#holly start
backend holly
       http-request replace-path /holly/(.*) /\1
       acl needs_scheme req.hdr_cnt(X-Scheme) eq 0
       http-request add-header X-Scheme https if needs_scheme { ssl_fc }
       http-request add-header X-Scheme http if needs_scheme !{ ssl_fc }
       http-request add-header X-Script-Name /holly
       server octoprint1 127.0.0.1:5001
       option forwardfor
#holly stop
#dolly start
backend dolly
       http-request replace-path /dolly/(.*) /\1
       acl needs_scheme req.hdr_cnt(X-Scheme) eq 0
       http-request add-header X-Scheme https if needs_scheme { ssl_fc }
       http-request add-header X-Scheme http if needs_scheme !{ ssl_fc }
       http-request add-header X-Script-Name /dolly
       server octoprint1 127.0.0.1:5002
       option forwardfor
#dolly stop
#stella start
backend stella
       http-request replace-path /stella/(.*) /\1
       acl needs_scheme req.hdr_cnt(X-Scheme) eq 0
       http-request add-header X-Scheme https if needs_scheme { ssl_fc }
       http-request add-header X-Scheme http if needs_scheme !{ ssl_fc }
       http-request add-header X-Script-Name /stella
       server octoprint1 127.0.0.1:5003
       option forwardfor
#stella stop
#cam_dolly start
backend cam_dolly
            http-request replace-path /cam_dolly/(.*)   /\1
            server webcam1 127.0.0.1:8002
#cam_dolly stop