paukstelis/octoprint_deploy #178: all instances are in failed state after upgrading to ubuntu 24.04

gkahansyto, 2024-10-24:
all my instances are in failed state after upgrading to ubuntu 24.04.  It seems to be related to python 3.12.  Any idea how to fix this.

paukstelis, 2024-10-24:
It is indeed the issue. It is a bit pathetic that Canonical doesn't have python 3.10/11 available in the the repos.

You will have to reinstall plugins (though all your plugin settings will be retained):
`python3 -m venv /home/$USER/OctoPrint`
`/home/$USER/OctoPrint/bin/pip install --upgrade pip`
`/home/$USER/OctoPrint/bin/pip install wheel`
`/home/$USER/OctoPrint/bin/pip install OctoPrint`

gkahansyto, 2024-10-24:
awesome! that works. thank you so much.

dgvigil, 2024-12-02:
Worked for me too. Thank you!

vagabond235, 2025-10-23:
Same issue on Linux Mint when upgrading to 22.2 - that solved it! hooray and thank you!