paukstelis/octoprint_deploy #180: Octoprint is not accessible from local network

den1251, 2024-11-09:
Devices on local network fail to connect to octoprint (using octoprint deploy). Can it be due to fulewall? Does octoprint deploy sets up firewall rules automatically? If I doesn't, what rules do I need to add?

System is CachyOS(arch based). Firewall if ufw.

den1251, 2024-11-10:
Maybe I don't have enough knowlege (correct me if anything wrong), but I don't think that disabling firewall entirely will not harm security. I meant that I want to connect to octoprint ui from other devices on the same network. Not printer directly etc. Don't I need to open just port 80?

paukstelis, 2024-11-10:
A firewall does little to nothing if you are only on a local network, if you trust the devices on that network.
If you used haproxy, then you can just open port 80. octoprint_deploy itself does nothing with firewalls.

den1251, 2024-11-13:
Allowing incoming connections for port 80 helps

closed 2024-11-13 by den1251