paukstelis/octoprint_deploy #186: Updating system/packages breaks all instances

den1251, 2025-01-01:
Updating packages/system on both arch `sudo packman -Syu` and fedora (using GUI) breaks all instances. Can we get this fixed or at least how to correctly update the system, when octoprint deploy is deployed?

paukstelis, 2025-01-01:
This has nothing to do with octoprint_deploy specifically. Nearly always it is an update to the python version that breaks things. Either don't upgrade python, or plan on reinstalling it after an update. ubuntu instructions here: https://github.com/paukstelis/octoprint_deploy/issues/178#issuecomment-2435425165

den1251, 2025-01-01:
Is it able to load all the dependencies specifically for octoprint, like flatpaks do?