paukstelis/octoprint_deploy #190: installing other python libraries

oldmanbluntz, 2025-02-04:
hello. I have been banging my head against the wall regarding this issue.

How do i install other python libraries and have them work with the instances of octoprint made by octoprint_deploy?

I use octoprint on a desktop, not a raspberry pi. I have Ubuntu 24.10 installed. I use an FTDI FT232H board to add GPIO to my desktop PC through USB. I have an octoprint plugin that is dependent on this working. This all works when i manually install octoprint. octoprint recognizes the board and the drivers, recognizes the SHIM i wrote to get rPi.GPIO functions to call to the FT232H using circuitpython libraries, and i can turn on and off the relays attached to the FT232H. the system recognizes the board, as i can access and use it through the python3 REPL in terminal.

i cannot get it to work with an instance of octoprint installed through octoprint_deploy.  My plugin will not install properly and will not work properly. 

I am trying to use octoprint_deploy because all of the previous methods used to automate the octoprint.service at startup don't seem to work in ubuntu 24.10, and i cannot for the life of me find any information on how to get it to start at system boot on ubuntu 24.10. it works with octoprint_deploy though. 

do you have any information that may help me on this? I know it's kind of a niche use case, but i would like to use octoprint_deploy.

paukstelis, 2025-02-05:
You need to install them in the OctoPrint virutal environment, typically with pip:
`$HOME/OctoPrint/bin/pip install <packagename>`

oldmanbluntz, 2025-02-05:
i tried installing them in the virtual environment. i guess i must not have been in it properly. 

can you not access the venv for octoprint from octoprint deploy by using the "source OctoPrint/bin/activate" command?

paukstelis, 2025-02-05:
You can do it that way. Running any of the python executables directly in ~/OctoPrint/bin/ will also use the venv.

oldmanbluntz, 2025-02-05:
yeah, that just does not seem to work. i don't know if it's changes between ubuntu 22.04 and 24.10, or changes between python 3.10 and python 3.12, but i just cannot seem to get this plugin to install anymore on octoprint. I can verify the drivers are installed in the venv for octoprint. but for some reason octoprint just will not install the plugin. and now it won't even work when i install octoprint manually. so i'm guessing its some of the changes to ubuntu 24.10 or python 3.12. the tutorial that i normally used to install the circuitpython stuff for the FT232h doesn't work the same now anyway. screaming at me when i try to pip3 install stuff not in a venv. it used to work in python 3.10 and ubuntu 22.04. 

It's all just gotten me absolutely furious. and how the octoprint team now just relies on octoprint-deploy and half of their damn tutorials are outdated regarding installing octoprint on linux........

all these headaches just to fix a damn lag issue with my cameras only in the octoprint UI every time the octoprint UI updated........

oldmanbluntz, 2025-02-05:
i think it's the changes in Ubuntu and how it's so closed up about python now....

I'm going to try ubuntu 22.04 again...... install ubuntu for the 12th time in 2 days......