paukstelis/octoprint_deploy #191: how to start it without adding printer?

KubaMiszcz, 2025-02-05:
I've run deploy.sh script without errors but cant start service
```
octo@OctoPrint:~$ sudo systemctl status octoprint
Unit octoprint.service could not be found.
```

there is no `octoprint/Octoprint/scripts/` to copy to initd

and when i try `http://192.168.x.x` it show 503: service unavailable

paukstelis, 2025-02-05:
You don't need to do any of that. That is the point of the script.
What did you name your first instance? You should be able to access it at http://192.168.x.x.:5000
`cat /etc/octoprint_deploy` will show you what instances you have created.

KubaMiszcz, 2025-02-05:
sure it is;]
it is output from 
```
octo@OctoPrint:~$ cat /etc/octoprint_deploy
haproxy: true
streamer: ustreamer
octoexec: /home/octo/OctoPrint/bin/octoprint
octopip: /home/octo/OctoPrint/bin/pip
octo@OctoPrint:~$
```

looks like I didnt create an instance
```
1) Config Only
2) Config and Gcode
Select what components of the template to copy: 1'
invalid option 1'
Select what components of the template to copy: 1
Ready to begin instance creation? [y/n]: y
Selected port is: 5000
Your new OctoPrint instance will be installed at /home/octo/.OctoPrint

Printer auto-detection must be completed before an instance can be created.
Begin printer auto-detection for udev entry? [y/n]: y

dmesg: klogctl failed: Operation not permitted
Plug your printer in via USB now (detection time-out in 1 min)
dmesg: read kernel buffer failed: Operation not permitted
dmesg: read kernel buffer failed: Operation not permitted
dmesg: read kernel buffer failed: Operation not permitted
...
...
```

is it possible without printer connected?

paukstelis, 2025-02-05:
You need to actually install an instance at the time of installing OctoPrint.
`sudo octoprint_deploy/octoprint_deploy.sh remove`
Then start the Prepare System again and go through until AFTER it creates an instance (answer yes and let it timeout when it tries to detect a printer).